projectacrn/acrn-hypervisor #144: Update README

shrmrf, 2018-05-05:
We should update the README file with a **Quickstart**, **license**  and **Contribution Guideline** section. 

[I have added](https://github.com/projectacrn/acrn-hypervisor/pull/143#issue-186149893) easy setup files in [this issue](https://github.com/projectacrn/acrn-hypervisor/pull/143#issue-186149893). If that issue is somewhat acceptable, I can expand and contribute to the README.

### Goals
- Use standard tools (no handwritten `bash` scripts)
- Use idioms provided by `docker` e.g. `.env` file to pass some environment variables
  -- This will assist in passing proxy environment variables in corporate settings

gvancuts, 2018-05-06:
Hi @shrmrf , all documentation should be put in the https://github.com/projectacrn/acrn-documentation repo so we can easily publish it on the [documentation website](https://projectacrn.github.io/latest/index.html). I would love to have you contribute to what we have there so feel free to create pull requests wherever you see an opportunity to bring more clarity to the docs we have!!

shrmrf, 2018-05-08:
@dbkinder exactly... I was going to push for those next.

gvancuts, 2018-05-08:
> I just realized that the license header is some proprietary license 

Not sure why you're saying that. This is the standard [BSD-3-Clause license](https://opensource.org/licenses/BSD-3-Clause) text. Am I missing something?

gvancuts, 2018-05-08:
👍 on all other suggestions made here :) (and thanks for clarifying your suggestion @shrmrf , it makes complete sense to me now)

shrmrf, 2018-05-08:
>Not sure why you're saying that. This is the standard BSD-3-Clause license text. Am I missing something?

Woops, my bad. I hadn't read the BSD licenses before. But still it looks more like C source code than a proper LICENSE file that Github likes :)

dbkinder, 2018-05-08:
GitHub is clever when you try to create a new file called LICENSE (and you have the right permissions) -- it will prompt you to pick one of the standard licenses and grab that for you, no typing needed.  That's how the MIT License on NVIDIA's dfc project, noted above, looks nice and pretty.

PR #170 (2018-05-11): doc: add project README
Simplify the previous README and add more information about project
resources.

Fixes: #144
commit 5b62cd27 (2018-05-11): doc: add project README

Simplify the previous README and add more information about project
resources.

Fixes: #144
closed 2018-05-11 by dbkinder