projectacrn/acrn-hypervisor #145: Enable Travis CI

shrmrf, 2018-05-05:
We should enable Travis CI and add a build-badge to the README file.

gvancuts, 2018-05-06:
There is a Jenkins service enabled now. The results of the test builds are displayed on the Pull Requests that are made. One thing we need to look at is making it more broadly visible. Today, only the test results are published and it's not possible to check the details.

I'm thinking that having Jenkins makes the need for Travis obsolete, would you agree?

shrmrf, 2018-05-07:
It's up to the repo owners to decide... You can close this issue then if you think it's not possible.

My preference for Travis comes from looking at all the other open source projects and Travis' integration with Github. Those badges look great on the README and logs can be verified by anyone in the world.

shrmrf, 2018-05-08:
Now I can't figure out why my pull request is failing https://github.com/projectacrn/acrn-hypervisor/pull/143

shrmrf, 2018-05-10:
We should close this issue if Travis CI isn't going to be enabled. 

I think it still makes sense to enable it though: this way the OSS community can contribute to the project.

jren1, 2018-05-28:
@shrmrf Recently the Travis is always reporting the compiling failures. Could you help fix the travis system?

shrmrf, 2018-05-28:
@jren1 sure... I'll take a look when i get back home (or during lunch break at work)... but can you point me to the Travis build logs you're looking at for example? I don't think Travis is enabled on this repo.. or?

gvancuts, 2018-05-28:
Hi @jren1 , we never created a top-level `.travis.yaml` file when we merged the `acrn-devicemodel` and `acrn-documentation` into this one. Could it be what's causing the failures you're seeing?

@shrmrf is right that we never enabled Travis on this repo so I guess you're looking at Travis testing out patches on your fork?

We'd need to update the set-up to enable Travis (inc. the `Dockerfile` since there are additional build dependencies required to build the codebase).