projectacrn/acrn-hypervisor #150: Move documentation to acrn-documentation repo

dbkinder, 2018-05-07:
Project documentation should be maintained in a central location rather than within each repo, so incorporate the contents of the /Documentation folder into the acrn-documentation repo and remove the /Documentation folder from the acrn-hypervisor folder.

gvancuts, 2018-05-15:
@YadongQi @mlouielu , we need to move the https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/Documentation/Trusty.txt document into the top-level doc folder and also turn it into a `*.rst` (Restructured Text file) in the process. I see both of you have contributed to the document, can you help us do that move? In case you don't know the ReST format, @dbkinder and I can help with that part but please confirm that the content is still accurate.

@dbkinder , where would you put this? I'm thinking in the "Developer Primer" section, what do you say?

YadongQi, 2018-05-15:
@gvancuts, Actually, I am not familiar with ReST format. I will really appreciate if you can help move  and reformat the document. Please add me to the reviewer list if you need my confirmation of the content. Thanks very much.

PR #547 (2018-07-05): Documentation: move Trusty documentation to doc/
Move the existing Trusty document to the doc/ folder (where
it belongs) and convert the text to ReST.

The Documentation/ folder under hypervisor/ is removed as all
documents should be put under doc/.

All technical information has been preserved or was already
available in other documents.

Fix #150
closed 2018-07-05 by dbkinder