projectacrn/acrn-hypervisor #165: Build a single hypervisor binary that works between different platforms

bryteise, 2018-05-11:
Rather than have a unique binary per PLATFORM configuration, enable the hypervisor to run regardless of which platform it is used on via run time detection and handling.

gvancuts, 2018-05-14:
Hi @bryteise , the idea is interesting but is that even doable? We're talking about two different executable formats, an EFI executable or one that can be launched by SBL. Isn't this akin to asking for a single executable that can be run on Windows or Linux?

bryteise, 2018-05-14:
My understanding that today SBL can launch a linux kernel itself so if that is true I am not sure why this isn't doable.

gvancuts, 2018-05-15:
I'll leave it to the expert but this would indeed be cool if we could have a single binary that supports both.

zhujun74, 2018-06-17:
i think it should be understood that ACRN very concerned about CPU arch and SOC memory remapping.so platform is good idea.

bryteise, 2018-06-17:
@zhujun74 Does that mean you think building multiple versions of ACRN for each platform is the best approach?

mattcurf, 2018-06-21:
As a contrary point of view, I desire a hypervisor that boots as fast as possible on a single platform by removing run time detection and handling, and hard coding as much as possible.

zhujun74, 2018-06-22:
Actually.As a SOC hyperviser it must be some configueable IO depend on system requirements.So does that meaning if it is possible keeping some freedom on configrable API for different requirements?

mattcurf, 2018-06-22:
@zhujun74 I agree for need for configuration to support a wide range of devices; I'm just suggesting that this configuration be done at compile time and not at run time.  In other words, source level configuration and support for a broad swarth of devices is good, but it is preferred that a single compiled binary boot as fast as possible on a single configured device.  In summary, my feedback is please don't provide flexibility at the expense of additional boot time or performance.  Thanks :)

mattcurf, 2018-06-22:
or ... provide the option for one configuration to boot dynamically for development purposes, but also provide the option for locking down configuration statically at compile time when it comes time to deploy a device broadly.

bryteise, 2018-06-22:
If there needs to be otherwise not required calls to probe hardware in order to handle a single run-time for all platforms, I'm very sympathetic to compile time configuration. It would be good to have data on a single-run time's performance impact.

zhujun74, 2018-06-23:
Option during the compiling time at source level is better.and given minmized time－boot setting by default board

closed 2019-03-25 by Mingyuan18