projectacrn/acrn-hypervisor #166: Errors in hypercall.h API documentation

dbkinder, 2018-05-11:
Two problems:

1. doxygen comments desribing `hcall_get_cpu_pm_state` are wrong (looks like a copy/paste from
   another function).  Needs a correct description of what it does, and description of each parameter. 
   Also, delete the blank line after the comment and the function prototype.  Here's how it currently 
   appears:

```
/**
 * @brief Switch VCPU state between Normal/Secure World.
 *
 * @param vcpu Pointer to VCPU data structure
 *
 * @return 0 on success, non-zero on error.
 */

int64_t hcall_get_cpu_pm_state(struct vm *vm, uint64_t cmd, uint64_t param);
```

2. The for the next function ( `hcall_world_switch` ), the vcpu parameter doc needs to be spelled correctly (change `@param VCPU` to `@param vcpu`  )

jren1, 2018-05-14:
Jason, Could you please look at that issue?

dbkinder, 2018-05-21:
@JasonChenCJ  @jren1  Can you fix the doxygen comments please.

dbkinder, 2018-05-23:
fixed by #236 and #247

closed 2018-05-23 by dbkinder