projectacrn/acrn-hypervisor #167: GitHub profiles must include real user name

dbkinder, 2018-05-11:
`git blame` shows the GitHub user name for the submitter (e.g., David B. Kinder).  The GitHub interface uses the GitHub name for contacting (e.g., @dbkinder) or assigning reviewers or issues.  If the user's profile does not list the real user name, there's no way to look up a person to make an assignment (e.g., what's the GitHub id for "Eddie Dong" or "Qi Yadong"), or who is id "ailun258"?

Please add your full name to your GitHub profile.  That way, when you try to assign Assignees or Reviewers, you can search for the person by name and their GitHub id will show up.

dbkinder, 2018-05-22:
Please add your full name to your GitHub profile: @Rraghuveer @feijiang1 @lifeix @lihao-intel @fuzhongl @guoch2 @gzhai7 @jwagantall @mingqiangchi @minhe1 @mwang106 @NanlinXie @qianouyang @shuox @yakuizhao @yechunliang

gvancuts, 2018-06-08:
Thank you to all of those that have updated their profile! 
@vgjayaku @shuox @mingqiangchi @minhe1 @feijiang1 @lihao-intel @mwang106 @NanlinXie 

@Rraghuveer  @lifeix  @fuzhongl @guoch2 @gzhai7 @jwagantall 
 @qianouyang @yakuizhao : can you please all do the same and update your Github profile with your real name. Thanks!

gvancuts, 2018-11-26:
@Rraghuveer @lifeix @fuzhongl @guoch2 @gzhai7 @jwagantall
@qianouyang @yakuizhao  @kromli, can you please all update your Github profiles to show your real name (and make your Intel email address public).

Mingyuan18, 2019-04-12:
Close it. Not a issue.

closed 2019-04-12 by Mingyuan18