projectacrn/acrn-hypervisor #175: Any UOS/VM can destroy other VM?

bzhu5, 2018-05-14:
https://github.com/projectacrn/acrn-hypervisor/blob/bbac18cf190f532ea3e7b56380219678b022f4c9/hypervisor/common/hypercall.c#L196

It seems like there are no restrictions for this hypercall (may have others), for example, can we check/test these?

- Can this hypercall be called from other guest VM?
- What if a ring-3 hypercall from a malicious application in guest VM is called to destroy other guest VM or itself?

bzhu5, 2018-05-14:
anyone knows how to "label" an issue? for example, I'd like to label this w/ "security"

gvancuts, 2018-05-14:
> anyone knows how to "label" an issue? for example, I'd like to label this w/ "security"

@Rraghuveer should be able to help.

bzhu5, 2018-05-15:
thank you @gvancuts !

@Rraghuveer is this because that i don't have permission to do that? it seems that I didn't find UI interface to allow me to either create a new label or label an issue with an existing label.  thank you.

dbkinder, 2018-05-15:
I went and created some labels you can choose from.

bzhu5, 2018-06-06:
looks like bug fixed. close it

closed 2018-06-06 by bzhu5