projectacrn/acrn-hypervisor #305: Compilation error on Fedora (all versions) and Ubuntu 18.04

gvancuts, 2018-05-29:
The `devicemodel` code will not compile on the following development distributions:
Note that the CI system will fail because there is a compilation error on the following distributions:
- Fedora 26, 27 and 28
- Ubuntu 18.04

Others distributions such as Ubuntu 14.04, 16.04 and Clearlinux are fine.

Here is the compilation error:
```
[ ! -e /root/acrn/build/devicemodel/hw/pci/virtio/rpmb_sim.o ] && mkdir -p /root/acrn/build/devicemodel/hw/pci/virtio/; \
cc -g -O0 -std=gnu11 -D_GNU_SOURCE -DNO_OPENSSL -m64 -Wall -ffunction-sections -Werror -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -fno-strict-aliasing -I/root/acrn/devicemodel/include -I/root/acrn/devicemodel/include/public -fstack-protector-strong -c hw/pci/virtio/rpmb_sim.c -o /root/acrn/build/devicemodel/hw/pci/virtio/rpmb_sim.o
hw/pci/virtio/rpmb_sim.c: In function 'rpmb_mac':
hw/pci/virtio/rpmb_sim.c:77:21: error: storage size of 'hmac_ctx' isn't known
  struct hmac_ctx_st hmac_ctx;
                     ^~~~~~~~
hw/pci/virtio/rpmb_sim.c:79:2: error: implicit declaration of function 'HMAC_CTX_init'; did you mean 'HMAC_CTX_new'? [-Werror=implicit-function-declaration]
  HMAC_CTX_init(&hmac_ctx);
  ^~~~~~~~~~~~~
  HMAC_CTX_new
hw/pci/virtio/rpmb_sim.c:106:2: error: implicit declaration of function 'HMAC_CTX_cleanup'; did you mean 'HMAC_CTX_get_md'? [-Werror=implicit-function-declaration]
  HMAC_CTX_cleanup(&hmac_ctx);
  ^~~~~~~~~~~~~~~~
  HMAC_CTX_get_md
hw/pci/virtio/rpmb_sim.c:77:21: error: unused variable 'hmac_ctx' [-Werror=unused-variable]
  struct hmac_ctx_st hmac_ctx;
                     ^~~~~~~~
cc1: all warnings being treated as errors
Makefile:164: recipe for target '/root/acrn/build/devicemodel/hw/pci/virtio/rpmb_sim.o' failed
make[1]: Leaving directory '/root/acrn/devicemodel'
make[1]: *** [/root/acrn/build/devicemodel/hw/pci/virtio/rpmb_sim.o] Error 1
Makefile:26: recipe for target 'devicemodel' failed
make: *** [devicemodel] Error 2
```

gvancuts, 2018-05-29:
@yang8621 @yingbinx , is this one for you?

yang8621, 2018-05-30:
@gvancuts  
Can you try to sudo apt-get update && apt-get install libssl-dev ?

gvancuts, 2018-05-30:
> Can you try to sudo apt-get update && apt-get install libssl-dev ?

It's already installed. Here is what is installed on the dev environment: https://github.com/gvancuts/acrn-dockerfiles/blob/master/Dockerfile.ubuntu18.04#L6-L14

gvancuts, 2018-05-30:
I found out the root cause of this issue. The code we have is not compatible with OpenSSL 1.1.x but works fine with OpenSSL 1.0.x. This is why it fails to compile on Fedora 26, 27 and 28 as well as Ubuntu 18.04. They all come with OpenSSL 1.1.x. All other distros I've tested (Ubuntu 14.04, 16.04 and Clearlinux).

@miguelinux , do you know if Clearlinux is planning to upgrade OpenSSL to version 1.1.x any time soon?

PR #445 (2018-06-19): DM: Fix build failure with Openssl 1.1+
Some APIs of Openssl 1.0 are deprecated in Openssl 1.1+.
Two different API implementations are adaptable to both
Openssl 1.1- and 1.1+.

Fixes: #305

Acked-by: Eddie Dong <eddie.dong@intel.com>
commit 18c355f6 (2018-06-19): DM: Fix build failure with Openssl 1.1+

Some APIs of Openssl 1.0 are deprecated in Openssl 1.1+.
Two different API implementations are adaptable to both
Openssl 1.1- and 1.1+.

Fixes: #305

Acked-by: Eddie Dong <eddie.dong@intel.com>
closed 2018-06-19 by lijinxia