projectacrn/acrn-hypervisor #334: Document the hypervisor commands available

gvancuts, 2018-05-31:
We need to document the various commands that are available from the hypervisor shell, e.g.: `list_vm`, `list_vcpu`, `pause_vcpu`, `resume_vcpu`, `reboot`, `set_loglevel`, `get_loglevel`, etc.

@ailun258 is this something you can help with?
@dbkinder where should we put this? (I'm actually think we should have a "User Documentation" section that includes the Getting Started Guides and other reference documents for the tools - thoughts?)

ailun258, 2018-06-01:
sure, we can do it, like i mentioned in our staff meeting,  at this moment, we really need to have documentations to describe how to use and play with this open source project,  i will create one later and submit a PR.

gvancuts, 2018-06-01:
Thanks @ailun258

zhujun74, 2018-06-19:
I Think it is important for ACRN early stage.some necessary DOC should be put on white－board together.Esp for memory mapping adress devidation setting.

gvancuts, 2018-11-23:
The [ACRN shell](https://projectacrn.github.io/latest/user-guides/acrn-shell.html) guide is available online. Closing this.

closed 2018-11-23 by gvancuts