projectacrn/acrn-hypervisor #424: ClearLinux desktop GUI of SOS fails to launch

kiendinh, 2018-06-14:
I set up the ACRN environment on the APL NUC and UP2 board with ClearLinux 22950. SOS was confirmed to boot up on ACRN. After installing the desktop GUI on ClearLinux following [this guide](https://clearlinux.org/documentation/clear-linux/guides/maintenance/enable-user-space#enable-user-space), the GUI failed to launch on the SOS in both HW. It could be launched after booting to the native kernel in both HW though.

Here are some logs of the gdm service after launching it with "systemctl start gdm" on SOS.
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (II) systemd-logind: got resume for 13:68
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (II) systemd-logind: got resume for 13:65
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (II) systemd-logind: got resume for 226:0
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (II) AIGLX: Resuming AIGLX clients after VT switch
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (EE) modeset(0): failed to set mode: Invalid argument
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (EE)
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: Fatal server error:
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (EE) EnterVT failed for screen 0
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (EE)
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: Please consult the The X.Org Foundation support
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]:          at http://wiki.x.org
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]:  for help.
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (EE)
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (II) AIGLX: Suspending AIGLX clients for VT switch
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[2315]: (EE) Server terminated with error (1). Closing log file.
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f gdm[329]: GLib: g_variant_new_string: assertion 'string != NULL' failed
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f gdm[329]: GLib: g_variant_new_string: assertion 'string != NULL' failed
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f gdm[329]: Child process -2315 was already dead.
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f gdm[329]: Child process 2311 was already dead.
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f gdm[329]: Unable to kill session worker process
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f systemd[1]: Started Session c6 of user gdm.
Jun 14 09:10:07 clr-144fdf96a7c1402c9b0ace212fae279f systemd-logind[256]: New session c6 of user gdm.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f gdm[329]: GdmDisplay: display lasted 0.710015 seconds
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f gdm[329]: GLib: g_variant_new_string: assertion 'string != NULL' failed
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f gdm[329]: Child process -6584 was already dead.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f gdm[329]: Child process 6580 was already dead.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f gdm[329]: Unable to kill session worker process
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f systemd[1]: Started Session c7 of user gdm.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f systemd-logind[256]: New session c7 of user gdm.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (--) Log file renamed from "/var/log/Xorg.pid-6615.log" to "/var/log/Xorg.0.log"
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: X.Org X Server 1.20.0
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: X Protocol Version 11, Revision 0
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: Build Operating System: Linux 4.16.6-563.native x86_64
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: Current Operating System: Linux clr-144fdf96a7c1402c9b0ace212fae279f 4.14.48-51.pk414-sos #1 SMP PREEMPT Sun Jun 10 06:34:34 UTC 2018 x86_64
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: Kernel command line: pci_devices_ignore=(0:18:2) maxcpus=1 console=tty0 console=ttyS0 i915.nuclear_pageflip=1 root=/dev/sda3 rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.tsd_init=7 i915.tsd_delay=2000 i915.domain_plane_owners=0x011111110000 i915.enable_guc_loading=0 i915.enable_guc_submission=0 i915.enable_preemption=1 i915.context_priority_mode=2 i915.enable_gvt=1 hvlog=2M@0x1FE00000 hugepagesz=1G hugepages=2
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: Build Date: 12 May 2018  02:05:35PM
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]:  
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: Current version of pixman: 0.34.0
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]:         Before reporting problems, check http://wiki.x.org
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]:         to make sure that you have the latest version.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: Markers: (--) probed, (**) from config file, (==) default setting,
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]:         (++) from command line, (!!) notice, (II) informational,
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (==) Log file: "/var/log/Xorg.0.log", Time: Thu Jun 14 09:10:08 2018
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (==) No Layout section.  Using the first Screen section.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (==) No screen section available. Using defaults.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (**) |-->Screen "Default Screen Section" (0)
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (**) |   |-->Monitor "<default monitor>"
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (==) No monitor specified for screen "Default Screen Section".
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]:         Using a default monitor configuration.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (==) Automatically adding devices
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (==) Automatically enabling devices
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (==) Automatically adding GPU devices
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (==) Max clients allowed: 256, resource mask: 0x1fffff
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (WW) The directory "/usr/share/fonts/X11/misc/" does not exist.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]:         Entry deleted from font path.
Jun 14 09:10:08 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/X11/OTF/".
[... cut ...]
Jun 14 09:10:12 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (II) systemd-logind: got pause for 13:65
Jun 14 09:10:12 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (II) systemd-logind: got pause for 13:67
Jun 14 09:10:12 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (II) systemd-logind: got pause for 13:68
Jun 14 09:10:12 clr-144fdf96a7c1402c9b0ace212fae279f /usr/libexec/gdm-x-session[6613]: (II) systemd-logind: got pause for 226:0

gvancuts, 2018-06-14:
I think it may have to do with the introduction of GVT-g and plane ownership. I think @ttzeng may have looked at this already.

kiendinh, 2018-07-17:
It looks like the fb console has been disabled with parameter i915.enable_initial_modeset=1 in the acrn.conf. This is by design. The GUI can be set up to launch on boot-up. For gdm, please follow this [guide](https://clearlinux.org/documentation/clear-linux/guides/maintenance/enable-user-space) to enable graphic UI on boot-up.

closed 2018-07-17 by kiendinh