projectacrn/acrn-hypervisor #430: Where is project version maintained?

dbkinder, 2018-06-14:
The project version seems to be stored in at least two places: the Makefiles for hypervisor and devicemodel both define a MAJOR_VERSION, MINOR_VERSION, and RC_VERSION and the top-level Makefile does not.

The project version should be maintained in one place (maybe an include file called ``version.mk`` at the top level that each of the lower-level Makefiles include with ``include ../version.mk``).

zhujun74, 2018-06-17:
It is good ides.for reason of hyperviser cpncerned lots OS&Hardware.Specfic Plarform should be included.

jren1, 2018-06-19:
Thanks, good proposal. I will follow up.

mwang106, 2018-12-22:
Now project version maintained in acrn-hypervisor/VERSION

closed 2018-12-22 by mwang106