projectacrn/acrn-hypervisor #438: doc build dependencies missing in CI environment

dbkinder, 2018-06-15:
Building documentation was recently included into the CI tests, but the dependencies for document building were's set up in the CI environment.

```
make -C /workspace/acrn-hypervisor-premerge-build-nuc/acrn-hypervisor/doc html BUILDDIR=/workspace/acrn-hypervisor-premerge-build-nuc/acrn-hypervisor/build/doc
make[1]: Entering directory '/workspace/acrn-hypervisor-premerge-build-nuc/acrn-hypervisor/doc'
make[1]: [Makefile:52: html] Error 2 (ignored)

New errors/warnings found, please fix them:
==============================================

Warning: sphinx_rtd_theme missing. Use pip to install it.

Extension error:
Could not import extension breathe (exception: No module named 'breathe')
make[1]: Leaving directory '/workspace/acrn-hypervisor-premerge-build-nuc/acrn-hypervisor/doc'
```

Doc build dependencies are documented in https://projectacrn.github.io/latest/howtos/process/docbuild.html#installing-the-documentation-tools

Are the CI environment configuration files checked into GitHub somewhere? (They should)  How do we update the CI environment?

NanlinXie, 2018-12-03:
This issue is fixed now and CI could check document build.

NanlinXie, 2018-12-03:
@dbkinder , Could you help to close this issue?

dbkinder, 2018-12-04:
doc build is now part of PR checking by the CI, so we can close this issue.

closed 2018-12-04 by dbkinder