projectacrn/acrn-hypervisor #443: launch_UOS.sh acrn-dm  report  Unable to setup memory (12)

huangweller, 2018-06-18:
acrn hv : 287fc4cb7faacf1a1c92ec77cb42bb1fc1e42d22
acrn kernel: 3c8fa189db1088589bbdd3948792e21250db20d1

issue:
launch_UOS.sh acrn-dm  report  Unable to setup memory (12)

##HV log:

ACRN Hypervisor
calibrate_tsc, tsc_hz=1881600000
[3827968us][cpu=0][sev=2][seq=1]:HV version 0.1-rc5-2018-06-18 14:23:10-287fc4c-dirty DBG build by weller, start time 3821056us
[3840512us][cpu=0][sev=2][seq=2]:API version 1.0
[3845632us][cpu=0][sev=2][seq=3]:Detect processor: Intel(R) Atom(TM) Processor E3900 @ 1.90GHz
[3855104us][cpu=0][sev=2][seq=4]:hardware support HV
[3872256us][cpu=0][sev=2][seq=5]:Start VM0
ACRN:\>vm_console

##launch_UOS.sh log:
Launch clearlinux UOS
cat: /sys/class/net/ens4/address: No such file or directory
tap device existed, reuse acrn_tap_yocto
memsize 2048M
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get kernel path /home/root/clearlinux/bzImage
SW_LOAD: get bootargs root=/dev/vda2 rw rootwait noxsave maxcpus=3 nohpet console=hvc0   console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M   consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=0x00000C   i915.enable_hangcheck=0 i915.nuclear_pageflip=1
VHM api version 1.0
Unable to setup memory (12)
root@:Dom0 ~ $ free -lm
              total        used        free      shared  buff/cache   available
Mem:           7809        4152        3388          52         269        3335
Low:           7809        4421        3388
High:             0           0           0
Swap:             0           0

gvancuts, 2018-06-18:
@ailun258 , @ttzeng can you take a look? We may be missing a `cma=` argument to the Service OS kernel command-line here.

jren1, 2018-06-18:
We have retired the CMA, and turned to hugetlb already for the UOS memory management.
Please ：
1) send the commit IDs of your ACRN and kernel.
2) send your SOS /proc/cmdline
3) add "hugepagesz=1G hugepages=4" to the SOS cmdline to reserve memory for UOS by hugetlb.

ailun258, 2018-06-19:
yes, if use CMA, there still is memory allocation problem,   hugeTLB is working well

jren1, 2018-06-19:
@mingqiangchi 
     Please give the steps about how to configure hugetlb and configure UOS memory.
@ailun258 
    Please update the GSG accordingly.

mwang106, 2018-12-22:
GSG has updated this part.

closed 2018-12-22 by mwang106