projectacrn/acrn-hypervisor #454: update the SOS kernel ,then can not login from screen via hdmi interface

fangwei-yangzhou, 2018-06-19:
On my hand is up2 board, I install clearlinux(version: 21260). 
When I compiled the acrn-kernel based on the kernel_uefi_sos, then I copy the bzImage to EFI/org.clearlinux/kernel-org.clearlinux.pk414-sos.4.14.44-42. I reboot the board, then the SOS system bring up, the screen via hdmi interface show login window. When I enter username , the terminal hang, then throw out the information as below:
clr-531f8fbc1c854646bbf9cfa9704e23f0 login: [   31.075522] random: crng init done
[   31.075692] random: 7 urandom warning(s) missed due to ratelimiting

But I open the console terminal ,then enter "vm_console", in the SOS console ,I can enter the username and password ,then loggin in.

Before I update the SOS kernel, all is ok.

jren1, 2018-06-19:
@shiqingg  Can you please take a look?

mattcurf, 2018-06-21:
Is commit e3174bb50ac65c8c352c0d058a5d6e8394fe0767 responsible?

shiqingg, 2018-06-21:
I will try to reproduce this issue and get updated later.

ttzeng, 2018-06-22:
​Actually, the system doesn't hang, just the input switched to other
terminals. You could press ALT-CTRL-F2 to continue. FYI.

Regards,
Tonny​

shiqingg, 2018-06-22:
Just tried the latest kernel in github on UP2 board.
Keyboard is working fine with the HDMI display.

Could you please check your kernel version and make sure the corresponding kernel modules have copied into rootfs?
$ uname -a
$ ls /lib/modules/

mwang106, 2018-12-22:
@fangwei-yangzhou do you still hit this issue?

Mingyuan18, 2019-03-22:
Close this issue because there is no reply for a long time

closed 2019-03-22 by Mingyuan18