projectacrn/acrn-hypervisor #480: failed to build arch at ubuntu16.04 + gcc-5

zhijianli88, 2018-06-22:
try to bisect, and figure out the FBC
```
lizj@G08FNSTD140138:~/workspace/Audio/acrn-hypervisor$ git bisect log 
git bisect start
# good: [2922a657c919af0542a29b6655374287dd984008] hv: fix compile error
git bisect good 2922a657c919af0542a29b6655374287dd984008
# bad: [804026168c208077c00b058102b14cfa856e47de] HV: remove unused API lookp_entry_by_id()
git bisect bad 804026168c208077c00b058102b14cfa856e47de
# bad: [f52a25db7e10f575cf59282fd9d2a460b5ae3096] HV: ptdev: convert vectors in msi_info to unsigned integers
git bisect bad f52a25db7e10f575cf59282fd9d2a460b5ae3096
# bad: [3027bfab109c8f076c65765394b7b81b64c8cc07] HV: treewide: enforce unsignedness of pcpu_id
git bisect bad 3027bfab109c8f076c65765394b7b81b64c8cc07
# first bad commit: [3027bfab109c8f076c65765394b7b81b64c8cc07] HV: treewide: enforce unsignedness of pcpu_id
```
i use ubuntu 16.04 and gcc-5
```
lizj@G08FNSTD140138:~/workspace/Audio/acrn-hypervisor$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.9' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.9) 
```

zhijianli88, 2018-06-22:
attached the failures
···
[ ! -e /home/lizj/workspace/Audio/acrn-hypervisor/build/hypervisor/arch/x86/trampoline.o ] && mkdir -p /home/lizj/workspace/Audio/acrn-hypervisor/build/hypervisor/arch/x86/; \
cc  -Iinclude  -Iinclude/lib  -Iinclude/lib/crypto  -Iinclude/common  -Iinclude/arch/x86  -Iinclude/arch/x86/guest  -Iinclude/debug  -Iinclude/public  -Iinclude/common  -Ibsp/include  -Ibsp/uefi/include/bsp  -Iboot/include  -I/home/lizj/workspace/Audio/acrn-hypervisor/build/hypervisor/include -I. -m64 -nostdinc -nostdlib -gdwarf-2 -DASSEMBLER=1 -c arch/x86/trampoline.S -o /home/lizj/workspace/Audio/acrn-hypervisor/build/hypervisor/arch/x86/trampoline.o
arch/x86/trampoline.S: Assembler messages:
arch/x86/trampoline.S:81: Error: missing ')'
arch/x86/trampoline.S:81: Error: missing ')'
arch/x86/trampoline.S:81: Error: junk `U<<3)|(1U<<5)|(1U<<6)|(1U<<9)|(1U<<10))' after expression
arch/x86/trampoline.S:92: Error: junk `U' after expression
arch/x86/trampoline.S:94: Error: missing ')'
arch/x86/trampoline.S:94: Error: junk `U<<8)' after expression
arch/x86/trampoline.S:101: Error: missing ')'
arch/x86/trampoline.S:101: Error: missing ')'
arch/x86/trampoline.S:101: Error: junk `U<<31)|(1U<<0)|(1U<<1)|(1U<<5))' after expression
arch/x86/trampoline.S:146: Error: junk `UL' after expression
arch/x86/trampoline.S:150: Warning: indirect jmp without `*'
arch/x86/trampoline.S:183: Error: missing ')'
arch/x86/trampoline.S:183: Error: junk at end of line, first unrecognized character is `U'
arch/x86/trampoline.S:192: Error: missing ')'
arch/x86/trampoline.S:192: Error: junk at end of line, first unrecognized character is `U'
arch/x86/trampoline.S:192: Error: missing ')'
arch/x86/trampoline.S:192: Error: junk at end of line, first unrecognized character is `U'
...
arch/x86/trampoline.S:199: Error: missing ')'
arch/x86/trampoline.S:199: Error: junk at end of line, first unrecognized character is `U'
arch/x86/trampoline.S:199: Error: missing ')'
arch/x86/trampoline.S:199: Error: junk at end of line, first unrecognized character is `U'
Makefile:256: recipe for target '/home/lizj/workspace/Audio/acrn-hypervisor/build/hypervisor/arch/x86/trampoline.o' failed
make[1]: *** [/home/lizj/workspace/Audio/acrn-hypervisor/build/hypervisor/arch/x86/trampoline.o] Error 1
make[1]: Leaving directory '/home/lizj/workspace/Audio/acrn-hypervisor/hypervisor'
Makefile:20: recipe for target 'hypervisor' failed
make: *** [hypervisor] Error 2
···

junjiemao1, 2018-06-22:
Hi @zhijianli88 , thanks for reporting this. The root cause is the assembler not supporting C-style unsigned constants like 0xffU (refer to https://sourceware.org/binutils/docs/as/Integers.html#Integers for the syntax of integers supported by binutils as).

Though not officially supported in the manual, I find the assembler in binutils >= 2.28 understands such constants while binutils 2.26 (shipped with Ubuntu 16.04) does not. You can either upgrade binutils or pick #485 which fixes this issue.

PR #485 (2018-06-25): HV: treewide: fix C-style unsigned constants in assembly
MISRA C requires that unsigned constants should have the 'U' suffix, while this
C syntax is not accepted by binutils assembler per binutil manual.

This patch explicitly spells out the unsigned constants used in the assembly
files while tracking the original expressions in comments. This fixes build
failure when using binutils <= 2.26.

This patch fixes #480.

v2 -> v3:

    * Explicitly spell out the unsigned constants in assembly, instead of
      duplicating the macros in headers which break the integrity of the
      definitions.

v1 [... cut]
closed 2018-06-25 by lijinxia