projectacrn/acrn-hypervisor #511: sos bootargs: /proc/cmdline is different from dmesg and sos_bootargs_debug.txt

huangweller, 2018-06-27:
i found the boot args of sos from  sos_bootargs_debug.txt, sos proc/cmdline and dmesg are different.

**bootarg file : sos_bootargs_debug.txt
contents:**
pci_devices_ignore=(0:18:2) noxsave maxcpus=1 console=tty0 console=ttyS0 i915.nuclear_pageflip=1 root=/dev/mmcblk1p1 rw rootwait clocksource=hpet quiet loglevel=3 no_timer_check consoleblank=0 i915.enable_initial_modeset=1 i915.tsd_init=7 i915.tsd_delay=2000 video=DP-1:d  video=DP-2:d i915.avail_planes_per_pipe=0x00000F i915.domain_plane_owners=0x011111110000 i915.enable_guc_loading=0 i915.enable_guc_submission=0 i915.enable_preemption=1 i915.context_priority_mode=2 i915.enable_gvt=1 hvlog=2M@0x6de00000 cma=6144M@0x100000000

**after sos stat up,  the contents  /proc/cmdline  contain more information than the sos_bootargs_debug.txt defined.
cat /proc/cmdline** 
pci_devices_ignore=(0:18:2) noxsave maxcpus=1 console=tty0 console=ttyS0 i915.nuclear_pageflip=1 root=/dev/mmcblk1p1 rw rootwait clocksource=hpet quiet loglevel=3 no_timer_check consoleblank=0 i915.enable_initial_modeset=1 i915.tsd_init=7 i915.tsd_delay=2000 video=DP-1:d video=DP-2:d i915.avail_planes_per_pipe=0x00000F i915.domain_plane_owners=0x011111110000 i915.enable_guc_loading=0 i915.enable_guc_submission=0 i915.enable_preemption=1 i915.context_priority_mode=2 i915.enable_gvt=1 hvlog=2M@0x6de00000 cma=6144M@0x100000000 androidboot.brand=intel androidboot.name=broxton_p androidboot.model=gr_mrb androidboot.device=gr_mrb ABL.bdev=mmc2:/iasImage ABL.boot=1400 ABL.csever=3.1.50.2238 ABL.bpdt1=0x000055aa,0x00000000 ABL.bpdt2=0x000055aa,0x00000000 ABL.hwver=51,4,1900,b00f,0,8192 ABL.mrcthreshold=0,0 ABL.memser=0x00000001 ABL.reset=power-on ABL.seed=0,0 ABL.seed_list=0,0 ABL.timestamps=64@0xc0000 ABL.consbuf=0xdc000 ABL.version=rel.1804-Android ABL.secureboot=0 ABL.virtual_mode=1 ABL.boot_target=NORMAL_BOOT trusty.param_addr=0xc5636 androidboot.bootloader=rel.1804-Android androidboot.serialno=R1J56L4ddb9080 androidboot.bootreason=power-on pci_devices_ignore=(0:18:2) noxsave maxcpus=1 console=tty0 console=ttyS0 i915.nuclear_pageflip=1 root=/dev/mmcblk1p1 rw rootwait clocksource=hpet quiet loglevel=3 no_timer_check consoleblank=0 i915.enable_initial_modeset=1 i915.tsd_init=7 i915.tsd_delay=2000 video=DP-1:d  video=DP-2:d i915.avail_planes_per_pipe=0x00000F i915.domain_plane_owners=0x011111110000 i915.enable_guc_loading=0 i915.enable_guc_submission=0 i915.enable_preemption=1 i915.context_priority_mode=2 i915.enable_gvt=1 hvlog=2M@0x6de00000 cma=6144M@0x100000000

**boot args from dmesg**

root@:Dom0 ~ $ dmesg
[    0.000000] Linux version 4.14.41-sos+ (weller@workstation) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.9)) #1 SMP PREEMPT Mon Jun 18 23:09:54 CST 2018
[    0.000000] Command line: pci_devices_ignore=(0:18:2) noxsave maxcpus=1 console=tty0 console=ttyS0 i915.nuclear_pageflip=1 root=/dev/mmcblk1p1 rw rootwait clocksource=hpet quiet loglevel=3 no_timer_check consoleblank=0 i915.enable_initial_modeset=1 i915.tsd_init=7 i915.tsd_delay=2000 video=DP-1:d video=DP-2:d i915.avail_planes_per_pipe=0x00000F i915.domain_plane_owners=0x011111110000 i915.enable_guc_loading=0 i915.enable_guc_submission=0 i915.enable_preemption=1 i915.context_priority_mode=2 i915.enable_gvt=1 hvlog=2M@0x6de00000 cma=6144M@0x100000000 androidboot.brand=intel androidboot.name=broxton_p androidboot.model=gr_mrb androidboot.device=gr_mrb ABL.bdev=mmc2:/iasImage ABL.boot=1400 ABL.csever=3.1.50.2238 ABL.bpdt1=0x000055aa,0x00000000 ABL.bpdt2=0x000055aa,0x00000000 ABL.hwver=51,4,1900,b00f,0,8192 ABL.mrcthreshold=0,0 ABL.memser=0x00000001 ABL.reset=power-on ABL.seed=0,0 ABL.seed_list=0,0 ABL.timestamps=64@0xc0000 ABL.consbuf=0xdc000 ABL.version=rel.1804-Android ABL.secureboot=0 A
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:

jsun26intel, 2018-06-29:
SBL/ABL would pass bootargs also, so kernel cmdline includes stitched sos_bootargs.txt and bootargs info from SBL, this is by design.

mwang106, 2018-12-22:
Closed this issue as it is by design.

closed 2018-12-22 by mwang106