projectacrn/acrn-hypervisor #512: launch UOS fail

huangweller, 2018-06-27:
Coud you help to check why i got fail when launch UOS ?
i built the UOS from the acrn kernel.

**LOG**
root@:Dom0 ~ $ 
root@:Dom0 ~ $ ./launch_UOS.sh
clearlinux  lost+found
clearlinux.img
Launch clearlinux UOS
cat: /sys/class/net/ens4/address: No such file or directory
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get kernel path /home/root/clearlinux/bzImage
SW_LOAD: get bootargs root=/dev/vda2 rw rootwait noxsave maxcpus=3 nohpet console=hvc0   console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M   consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=0x00000C   i915.enable_hangcheck=0 i915.nuclear_pageflip=1
VHM api version 1.0
polling 1024...
Listening 1024...
***********************************************
virt-console backend redirected to /dev/pts/0
***********************************************
SW_LOAD: bootargs copied to guest 0x7fffe000
SW_LOAD: kernel /home/root/clearlinux/bzImage size 12411648 copied to guest 0x1000000
SW_LOAD: found setup sig @ 0x000042b8, setup_size is 0x00004400
SW_LOAD: build zeropage for bootargs addr: 0x7fffe000
SW_LOAD: build e820 6 entries to addr: 0x7fc8501b62d0
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000f0000,  type 0x1
SW_LOAD: entry[1]: addr 0x00000000000f0000, size 0x0000000000010000,  type 0x2
SW_LOAD: entry[2]: addr 0x0000000000100000, size 0x000000007ff00000,  type 0x1
SW_LOAD: entry[3]: addr 0x0000000080000000, size 0x0000000000000000,  type 0x2
SW_LOAD: entry[4]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[5]: addr 0x0000000100000000, size 0x0000000000100000,  type 0x2
SW_LOAD: zeropage prepared @ 0x7ffff000, kernel_entry_addr=0x1004400

gvancuts, 2018-06-28:
Hi @huangweller , what is your environment (i.e. ACRN version)? Your prompt makes it look like you're running Xen (`root@:Dom0`)?

Have you had any success using the 'vanilla' User OS (i.e. the one as described in our documentation)? I would like to understand if this is a problem with the kernel you've built or is something that was there all along.

huangweller, 2018-06-30:
hello @gvancuts 
the HV, kernel and DM are all come from the ACRN github.
"root@:Dom0" this is because i use a old rootfs which released by intel to customer before. it is built from yocto.

as i understand, the acrn-kernel can be built as SOS and UOS with different kernel configuration. but i found i can not launch the UOS successfully.
i will try 'vanilla' User OS

and thanks for your reply.

mwang106, 2018-12-22:
@huangweller can we close this issue?

gvancuts, 2019-03-05:
Closing this issue as there has been no activity for a while. @huangweller , please let us know if you are still encountering issues and we will look at those.

closed 2019-03-05 by gvancuts