projectacrn/acrn-hypervisor #524: Trying to setup ACRN with static IP address

bem-nrl, 2018-06-28:
Since I cannot use DHCP at my workplace, I had to follow these instructions to setup a static ip address:

https://clearlinux.org/documentation/clear-linux/reference/bundles/os-core

After doing this I was able to follow your instructions for setting up ACRN on my NUC here:

https://projectacrn.github.io/latest/getting_started/index.html#

I successfully booted up ACRN however, the SOS that booted lost my static IP connectivity and therefore I was unable to download the reference UOS via:

curl -O https://download.clearlinux.org/releases/22140/clear/clear-22140-kvm.img.xz

What is the recommended procedure to continue from this point in your instructions?

Many thanks.

gvancuts, 2018-06-28:
> I successfully booted up ACRN however, the SOS that booted lost my static IP connectivity

Could it be that the device name is different when running on ACRN? If so, perhaps you can simply adjust the /etc/systemd/network/50-static.network file (or better, add a new one so you have one for each name and you would have connectivity regardless of whether you're booting ACRN or native Clear Linux)

bem-nrl, 2018-06-29:
Great idea @gvancuts.  Unfortunately, the devices have the same name (enp3s0).

Booting up ACRN has left me with an enp3s0 that has no assigned IP address along with the pair of acrn devices that implement the acrn bridge.  Talk about a bridge to nowhere.

Any ideas on how to fix this would be much appreciated.

Thanks.

gvancuts, 2018-06-29:
I forgot about that bridge... I think the problem is that the bridge we set-up uses DHCP by default. Can you modify the `/usr/lib/systemd/network/50-eth.network` file (section `[Network]`) and put your static network parameters in there?

You could try then to:
```
# systemctl daemon-reload
# systemctl restart systemd-networkd
```
And see if it helps. If it does not, try with a real reboot and send us the the output of `journalctl -b -u systemd-networkd` if it's still not functional.

bem-nrl, 2018-06-29:
Awesome!  Works like a champ now.  Thanks @gvancuts !

bem-nrl, 2018-06-29:
Resuming the process now that the internet connectivity has been restored I attempt to launch the uos via the following script:

`
#!/bin/bash

function launch_clear()
{
vm_name=vm$1

#check if the vm is running or not
vm_ps=$(pgrep -a -f acrn-dm)
result=$(echo $vm_ps | grep "${vm_name}")
if [[ "$result" != "" ]]; then
  echo "$vm_name is running, can't create twice!"
  exit
fi

#for memsize setting
mem_size=1000M

# make sure there is enough 2M hugepages in the pool
echo 1024 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages

acrn-dm -T -A -m $mem_size -c $2 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  -s 2,pci-gvt -G "$3" \
  -s 5,virtio-console,@pty:pty_port \
  -s 6,virtio-hyper_dmabuf \
  -s 3,virtio-blk,/root/clear-22140-kvm.img \
  -s 4,virtio-net,tap0 \
  -k /usr/lib/kernel/org.clearlinux.pk414-standard.4.14.51-60 \
  -B "root=/dev/vda3 rw rootwait maxcpus=$2 nohpet console=tty0 console=hvc0 \
  console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M \
  consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=$4 \
  i915.enable_hangcheck=0 i915.nuclear_pageflip=1 i915.enable_initial_modeset=0" $vm_name
}

launch_clear 2 1 "64 448 8" 0x070F00 clear
`

with the following results:

`
root@clear ~ # ./launch_uos.sh
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get kernel path /usr/lib/kernel/org.clearlinux.pk414-standard.4.14.51-60
SW_LOAD: get bootargs root=/dev/vda3 rw rootwait maxcpus=1 nohpet console=tty0 console=hvc0   console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M   consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=0x070F00   i915.enable_hangcheck=0 i915.nuclear_pageflip=1 i915.enable_initial_modeset=0
VHM api version 1.0
open hugetlbfs file /run/hugepage/acrn/huge_lv1/D279543825D611E8864ECB7A18B34643

open hugetlbfs file /run/hugepage/acrn/huge_lv2/D279543825D611E8864ECB7A18B34643

level 0 free/need pages:524/500 page size:0x200000
level 1 free/need pages:0/0 page size:0x40000000

try to setup hugepage with:
	level 0 - lowmem 0x3e800000, highmem 0x0
	level 1 - lowmem 0x0, highmem 0x0
total_size 0x3ea00000

mmap ptr 0x0x7f01b91f5000 -> baseaddr 0x0x7f01b9200000
mmap 0x3e800000@0x7f01b9200000
touch 500 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x3e800000, highmem 0x0
	level 1 - lowmem 0x0, highmem 0x0
total_size 0x3ea00000

Listening 1024...
polling 1024...
/tmp/dm.XxneqOp     59:           0xFFFFFFFF,             // Range Maximum
                                                                          Error    6139 -     Constant out of range ^  (0xFFFFFFFF, allowable: 0x0-0xFFFF)

              /tmp/dm.XxneqOp     61:           0xFFFFE000,             // Length
                                                                                 Error    6139 -     Constant out of range ^  (0xFFFFE000, allowable: 0x0-0xFFFF)

                     mngr_client_new 121
Stop listening 1024...
Stop polling 1024...
`

I also attempted to verify whether any vm was launched and got the following results:

`
root@clear ~ # acrnctl list
find: ‘/opt/acrn/conf/add/’: No such file or directory
There are no VMs
`

Not sure if the No such file or directory above indicates a problem or not as the VM was never launched successful.

lyan3, 2018-12-25:
This issue has been fixed, may close now.

closed 2019-01-08 by mwang106