projectacrn/acrn-hypervisor #526: hypervisor information printed out when using virtio-console in launch android

yonghou2016, 2018-06-29:
in the launch_android script, using following code:

acrn-dm -A -m $mem_size -c $2 -G "$3" -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio 
-l com1,stdio
-s 5,virtio-console,@pty:pty_port \
-s 2,pci-gvt 
-s 9,virtio-net,$tap 
-s 4,virtio-blk,/data/$5/$5.img 
-s 7,passthru,0/15/0 
-s 8,passthru,0/15/1 
-s 10,virtio-hyper_dmabuf 
-s 14,passthru,0/e/0 
-s 15,passthru,0/f/0 
-s 24,passthru,0/17/0 
-s 22,passthru,0/16/0 
-M 
-k /home/root/$5/boot.img-kernel -r /home/root/$5/boot.img-ramdisk.gz 
-B "initrd=0x48800000 root=/dev/ram0 rw rootwait noxsave maxcpus=$2 
nohpet console=ttyS0 no_timer_check ignore_loglevel 
log_buf_len=16M consoleblank=0 tsc=reliable reboot_panic=p,w 
intel_iommu=off relative_sleep_states=1 vga=current 
androidboot.hardware=gordon_peak_acrn firmware_class.path=/vendor/firmware 
androidboot.selinux=permissive sys.init_log_level=7 
androidboot.serialno=$ser 
i915.enable_rc6=1 drm.atomic=1 i915.nuclear_pageflip=1 i915.enable_fbc=1 
i915.modeset=1 drm.vblankoffdelay=1 i915.fastboot=1 i915.enable_guc_loading=0 
i915.hpd_sense_invert=0x7 i915.avail_planes_per_pipe=$4 
i915.enable_hangcheck=0 use_nuclear_flip=1 
firmware_class.path=/vendor/firmware" $vm_name

[pre-condition]: add one line : " -s 5,virtio-console,@pty:pty_port " like the above script
[issue]: the sos console will dump hypervisor log as below:

[112832768us][cpu=1][sev=1][seq=1947]:Error: Unhandled VM exit condition from guest at 0x00000000000009f5
[112843520us][cpu=1][sev=1][seq=1948]:Exit Reason: 0x0000000000000002
[112850688us][cpu=1][sev=2][seq=1949]:Exit qualification: 0x0000000000000000
[112858368us][cpu=1][sev=1][seq=1950]:Error: Unhandled VM exit condition from guest at 0x00000000000009f5
[112869120us][cpu=1][sev=1][seq=1951]:Exit Reason: 0x0000000000000002
[112876288us][cpu=1][sev=2][seq=1952]:Exit qualification: 0x0000000000000000
[112883968us][cpu=1][sev=1][seq=1953]:Error: Unhandled VM exit condition from guest at 0x00000000000009f5
[112894720us][cpu=1][sev=1][seq=1954]:Exit Reason: 0x0000000000000002
[112901888us][cpu=1][sev=2][seq=1955]:Exit qualification: 0x0000000000000000
[112909568us][cpu=1][sev=1][seq=1956]:Error: Unhandled VM exit condition from guest at 0x00000000000009f5
[112920320us][cpu=1][sev=1][seq=1957]:Exit Reason: 0x0000000000000002
[112927488us][cpu=1][sev=2][seq=1958]:Exit qualifica

jren1, 2018-06-29:
@mcao6 Please look at it.

mcao6, 2018-07-02:
ok, will check that, but from the script, it should not be related with virtio-console, for  console=ttyS0 even not use it, and also the Android has not boot up. So I doubt it is other issue. @yonghou2016 you can do the test to remove "-s 5,virtio-console,@pty:pty_port" as a comparison.

yonghou2016, 2018-07-02:
After remove it, Android  UOS can boot as normal.

regards
Yong

mcao6, 2018-07-02:
root       536 15.3  0.0 3549596 6636 pts/0    Sl+  22:20   0:04 acrn-dm -T -A -m 2048M -c 3 -s 2,pci-gvt -G 64 448 8 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio -l com1,stdio -s 5,virtio-console,@pty:pty_port -s 9,virtio-net,tap_AaaG -s 3,virtio-blk,b,/data/android/android.img -s 7,passthru,0/15/0 -s 8,passthru,0/15/1 -s 13,virtio-rpmb -s 10,virtio-hyper_dmabuf -s 11,wdt-i6300esb -s 14,passthru,0/e/0 -s 23,passthru,0/17/0 -s 15,passthru,0/f/0 -s 27,passthru,0/1b/0 -s 24,passthru,0/18/0 -s 12,passthru,0/3/0 -s 22,passthru,0/16/0 -i /run/acrn/ioc_vm1-13:ce:5d,0x20 -l com2,/run/acrn/ioc_vm1-13:ce:5d -M --vsbl /usr/share/acrn/bios/VSBL.bin --enable_trusty -B maxcpus=3 nohpet tsc=reliable intel_iommu=off    androidboot.serialno=R1J56L8006bb7b    i915.enable_rc6=1 i915.enable_fbc=1 i915.enable_guc_loading=0 i915.avail_planes_per_pipe=0x070F00    i915.enable_hangcheck=0 use_nuclear_flip=1 i915.enable_initial_modeset=1  vm1-13:ce:5d

mcao6, 2018-07-02:
I just checked on my MRB, it is OK to add the params: -s 5,virtio-console,@pty:pty_port. I'm not sure which android you're using.  If you add this param for Linux guest OS, it runs OK, it could be caused by android kernel configure.

mcao6, 2018-07-02:
@yonghou2016 could you offer the full log when dm launch android?

yonghou2016, 2018-07-02:
I used one “old” android kernel- 4.14.25 released by Intel. once I switch it to the latest android kernel , I will share you the result.

regards
Yong

chejianj, 2018-07-03:
Currently the physical uart is shared among HV and SOS.
HV log routed to console will be printed out ALWAYS if its severity <= console_loglevel
the default console_loglevel is 3, so pr_fatal/pr_acrnlog/pr_err will be printed out even in SOS.

you can try below steps:
ACRN:\>get_loglevel
console_loglevel: 3, mem_loglevel: 5
ACRN:\>set_loglevel 5
ACRN:\>vm_console 0

after you launch a guest, you can see log from hv like below:
[77905408us][cpu=1][sev=5][seq=61]:break_page_table, Breaking 1GB -->2MB vaddr=0x0
[77913856us][cpu=1][sev=5][seq=62]:break_page_table, Breaking 2MB -->4KB vaddr=0x0
[77922304us][cpu=1][sev=5][seq=63]:break_page_table, Breaking 1GB -->2MB vaddr=0x200000000
[77931520us][cpu=1][sev=5][seq=64]:break_page_table, Breaking 2MB -->4KB vaddr=0x200000000

mcao6, 2018-07-03:
I could mis-understood this question's point, I was checking the crash issue; it also cares the "console" output issue. For the SOS/HV uart console, it is sharing between them; if you launch UOS in it (it is same in SOS), also sharing with HV. So usually, we will ssh to login SOS, then launch UOS, so the log will not mix.

mcao6, 2018-12-25:
@yonghou2016 if you still have question for this issue, if no, we'll close this issue in the coming weeks; if yes, please let me know.

closed 2019-01-08 by mwang106