projectacrn/acrn-hypervisor #546: question: trusty world

helen-fornazier, 2018-06-29:
Hi,

I am studying about the project, and in this slides https://elinux.org/images/3/3c/ACRN-brief2.pdf page 3 "Architecture Overview", in the User OS there is the Android world and the Trusty world.

If I understand correct, to support the Trusty world some support in hardware is required. I would like to know what is the current support Acrn have for TEE execution (I couldn't find anything in the docs regarding this).

Thanks

gvancuts, 2018-06-29:
Hi @helen-fornazier , there is a doc that will be added soon (PR pending). You can already check it out here: https://github.com/projectacrn/acrn-hypervisor/pull/547 . Hopefully, it will answer some of your questions.

gvancuts, 2018-07-05:
Hi @helen-fornazier , the page dedicated to Trusty is now live at https://projectacrn.github.io/latest/developer-guides/trusty.html

Can you let me know if you still have questions or if we can close this issue?

Thanks

helen-fornazier, 2018-07-08:
I was wondering if it has any hardware requirements, similar to TrustedZone in the ARM world, but I guess with Intel this can be implemented using the usual paging mechanism.
Thanks for your help, closing this issue.

closed 2018-07-08 by helen-fornazier