projectacrn/acrn-hypervisor #560: Unable to launch UOS

bem-nrl, 2018-07-02:
Following instructions on https://projectacrn.github.io/latest/getting-started/apl-nuc.html I was able to boot up ACRN using a static IP address without loosing my connectivity in the SOS thanks to your help.

However, I am unable to launch a UOS now.

Here is the output when attempting to launch a UOS:

[image]

The launch script that was used is as follows:

[image]

Attempting to determine if any VM was launched I then did the following:

[image]

Not sure if the missing file error above is an indication of some other problem.

Please advise.

Thanks.

gvancuts, 2018-07-04:
@ttzeng @ailun258 , could you take a look at this issue please?

bem-nrl, 2018-07-12:
This problem persists. Is there no resolution?

Perhaps someone could  provide a launch_uos.sh script that is working with the latest version of ACRN?

Would using static IP addressing instead of DHCP cause any trouble with this?

Ideally, a working /dev/sda image for the latest ClearLinux/ACRN for the NUC would be awesome provided it either did NOT require network access, or used a static IP address.

ttzeng, 2018-07-12:
@bem-nrl, I used to use `mem_size=1G` in my launch script, you may give it a try. I don't think the static IP address matter, I can launch the UOS even the device does not connect to the network. Besides, what's your acrn.conf settings? I also have `hugepagesz=1G hugepages=4` in my acrn.conf. FYI.

Mingyuan18, 2018-12-21:
Can't reproduce this issue.

closed 2018-12-21 by Mingyuan18