projectacrn/acrn-hypervisor #654: The Dockerfile turns off security

fenrus75, 2018-07-10:
RUN git config --global http.sslVerify false

that's obviously never an acceptable solution for anything

gvancuts, 2018-07-10:
This is something we had to put this when playing with Shippable, else it failed to clone the repo. We do not actually use these Dockerfiles in any automated fashion. There may be individuals that use them (like me) but that line isn't needed for those. We can remove it, I'll prepare a patch.

gvancuts, 2018-07-11:
A little more background as to what led to this workaround. The original problem is that it is not possible to clone https://github.com/projectacrn/acrn-hypervisor from a Clear Linux container, you get the following error:
```
Cloning into 'acrn-hypervisor'...
fatal: unable to access 'https://github.com/projectacrn/acrn-hypervisor/': SSL certificate problem: unable to get local issuer certificate
```
I do not see this issue when using a bare-metal installation of Clear Linux. When looking at the differences, I noticed that the bare-metal installation had a `/var/cache/ca-certs/anchors/` folder that contains a number of certificate in it. That folder isn't present in the container. I believe this should be created by [clrtrust](https://github.com/clearlinux/clrtrust). I tried to generate such trust store manully inside the container but that failed with the following error message:
```
root@f0b406d8e1d5 /root/acrn # clrtrust list
/var/cache/ca-certs is not a trust store.  Use clrtrust generate to create the store.
root@f0b406d8e1d5 /root/acrn # clrtrust generate
mktemp: failed to create file via template ‘/var/lock/clrtrust.lock.XXXXXX’: No such file or directory
rm: cannot remove '': No such file or directory
Failed to acquire lock file: /var/lock/clrtrust.lock.
If no other clrtrust process is running, remove the lock file manually and try again.
```

@fenrus75 , should I report this on https://github.com/clearlinux/clrtrust/issues to get it looked at by the experts?

Note: other containers based on Fedora, Ubuntu, Debian, CentOS do not exhibit this problem.

jren1, 2018-07-11:
@gvancuts But I have can git clones https:// even in Docker Clear Linux from my side. And The travis build can also pass even if I remove your workaround. What about from your side?

gvancuts, 2018-07-11:
> @gvancuts But I have can git clones https:// even in Docker Clear Linux from my side.

That's interesting. Can you share your exact steps (and Dockerfile)? 

> And The travis build can also pass even if I remove your workaround. What about from your side?

Yes, this does not affect Travis because it never does a `git clone` from the container. The source code is checked out separately and then used in the container thanks to the `-v $PWD:/root/acrn` parameter. This is why this line can safely be removed. Shippable works differently and that line was added when I was experimenting with it.

PR #656 (2018-07-11): Dockerfile: remove security by-pass in Clear Linux Dockerfile
"RUN git config --global http.sslVerify false" was needed with
a cloud-based CI system else it failed to clone the repository.

The project does not use that cloud-based CI system and hence
this workaround (security by-pass) is not needed.

Fix #654
closed 2018-07-11 by jren1