projectacrn/acrn-hypervisor #659: don't force http.sslVerify to be false in ClearLinux docker

jren1, 2018-07-10:
In https://github.com/projectacrn/acrn-hypervisor/blob/master/.travis-dockerfiles/Dockerfile.clearlinux:
http.sslVerify is forced to be false.
That could introduce the security concern.
Please remove it and try different workaround:
Solution#1: (that will use the Clearelinux auto-proxy)
$ unset https_proxy

Solution#1:
 $ export https_proxy=http://<proxy>:913/

Solution#3:
$ git clone https://.... 
==>
$ git clone http://

jren1, 2018-07-11:
@gvancuts Please review the fix: https://github.com/projectacrn/acrn-hypervisor/pull/665

gvancuts, 2018-07-11:
This is a duplicate of #654 , I have submitted a PR (#656) to address this.

NanlinXie, 2018-12-21:
PR merged, close this issue.

closed 2018-12-21 by NanlinXie