projectacrn/acrn-hypervisor #663: Black Screen when booting NUC with Hypervisor

nwang-intel, 2018-07-11:
we found a black screen issue when boot up NUC with Hypervisor, after debugging we found is kind of ClearLinux issue. here is the work around 

1.	In order to successfully boot up SOS/UOS with display(otherwise there is NO display after boot up SOS), below steps are needed before rebooting native clear linux to SOS, 
1)	After launching native clear linux, MUST install bundle desktop
        $ swupd bundle-add desktop
2)	get IP for clearlinux, record IP for ssh login sos in case SOS display failed after sos booted
        $ ip addr
3)	Execute below commands to enable display
         copy weston.service(attached file, please unzip the file before use it)  to /lib/systemd/system/weston.service
[weston.zip](https://github.com/projectacrn/acrn-hypervisor/files/2182853/weston.zip)
        $ systemctl enable Weston

when booting up SOS
1)	press F7
2)	choose “ACRN Hypervisor"   (hypervisor name set by efibootmgr )
4)	select “The ACRN Service OS”
5)	then enter SOS

after booting up UOS, step 1)~3) are needed too for UOS display .

dbkinder, 2018-07-11:
@nwang-intel  The instructions say a ``weston.service`` file is attached, but there is no such file here.

puneetse, 2018-07-19:
Thanks for the workaround. It worked for me. 
 
One comment on this step:
> 3. Create the file /lib/systemd/system/weston.service with the following content:

I think  `/etc/systemd/system/weston.service` is a better place to put the custom unit file, because Clear Linux is stateless and /usr is assumed to be owned by the Clear OS. More information on this here: https://clearlinux.org/features/stateless

rustylynch, 2018-07-24:
I don't understand why you need to install the desktop bundle just to run a command line argument to find the IP address.  /usr/bin/ip is installed with os-core, so you already have that with the most basic native install and you just have to run it in the virtual terminal

ttzeng, 2018-07-25:
@rustylynch , install the desktop bundle is to use the graphics tools, as the console is not visible when the system boots with the hypervisor and `enable_initial_modeset=1`, so what @nwang-intel is proposing is to get the IP address from a **native** session, then reboot the system and SSH into the SOS to setup the weston.

commit cad8492a (2018-08-15): enable weston to fix: #663

add user and enable weston service to fix issue: #663
closed 2018-08-15 by dbkinder