projectacrn/acrn-hypervisor #673: Build error after adding 'U/UL' suffix for unsigned contant value

ttzeng, 2018-07-11:
Commit `f81fcf2` causes the following build error. Any changes required for the build environment? Thanks.

```cc  -Iinclude  -Iinclude/lib  -Iinclude/lib/crypto  -Iinclude/common  -Iinclude/arch/x86  -Iinclude/arch/x86/guest  -Iinclude/debug  -Iinclude/public  -Iinclude/common  -Ibsp/include  -Ibsp/uefi/include/bsp  -Iboot/include  -I/opt/mnt/acrn-hypervisor/build/hypervisor/include -I.  -m64 -nostdinc -nostdlib  -gdwarf-2 -DASSEMBLER=1 -c arch/x86/trampoline.S -o /opt/mnt/acrn-hypervisor/build/hypervisor/arch/x86/trampoline.o
arch/x86/trampoline.S: Assembler messages:
arch/x86/trampoline.S:167: Warning: indirect jmp without `*'
arch/x86/trampoline.S:197: Error: junk at end of line, first unrecognized character is `U'
arch/x86/trampoline.S:202: Error: junk at end of line, first unrecognized character is `U'
arch/x86/trampoline.S:210: Error: junk at end of line, first unrecognized character is `U'
arch/x86/trampoline.S:210: Error: junk at end of line, first unrecognized character is `U'
arch/x86/trampoline.S:210: Error: junk at end of line, first unrecognized character is `U'
arch/x86/trampoline.S:210: Error: junk at end of line, first unrecognized character is `U'
arch/x86/trampoline.S:211: Error: junk at end of line, first unrecognized character is `U'
Makefile:268: recipe for target '/opt/mnt/acrn-hypervisor/build/hypervisor/arch/x86/trampoline.o' failed
make[1]: *** [/opt/mnt/acrn-hypervisor/build/hypervisor/arch/x86/trampoline.o] Error 1
make[1]: Leaving directory '/opt/mnt/acrn-hypervisor/hypervisor'
Makefile:20: recipe for target 'hypervisor' failed
make: *** [hypervisor] Error 2
```

gvancuts, 2018-07-11:
What is your dev environment @ttzeng ?

There was something similar in the past that got fixed with #485 (issue was #480).

ttzeng, 2018-07-11:
@gvancuts I used to build ACRN binaries using a Docker container running stock Ubuntu 16.04. Yes, a similar build error #480 happened few weeks ago, and that one was fixed until the introduction of the commit `f81fcf2` today (concluded by bisect the issue). Just curious how this passes the Travis before merging?

gvancuts, 2018-07-11:
> Just curious how this passes the Travis before merging?

Travis is not enabled, the infra piece is there but the service is not active on the repo so no checks is done. We have a Jenkins CI system too but I do not know if it performs that kind of test (I believe it does not, @NanlinXie can confirm).

junjiemao1, 2018-07-11:
@ttzeng This is an issue specific to binutils <=2.28 while our CI uses containers with a newer version of the tools. The latest issue has been fixed by #667.

ttzeng, 2018-07-11:
@junjiemao1 thanks for cooking the fix, the current HEAD `5a7ccc8` back to work now.

closed 2018-07-11 by ttzeng