projectacrn/acrn-hypervisor #677: The SSD disk id in UOS is different with the number in SOS when pass-through SSD device to UOS by 00:15:0, and it will change after reboot.

nwang-intel, 2018-07-12:
repro steps 

1: Boot to SOS
 `root@acrn-xiangli02 ~ # ls /dev/disk/by-id/
ata-FORESEE_120GB_SSD_H46092R000318
 ata-FORESEE_120GB_SSD_H46092R000318-part1
 ata-KINGSTON_SA400S37120G_50026B7782008E43
 ata-KINGSTON_SA400S37120G_50026B7782008E43-part1
 ata-KINGSTON_SA400S37120G_50026B7782008E43-part2
 ata-KINGSTON_SA400S37120G_50026B7782008E43-part3
 mmc-SC32G_0x490d2b8c
 mmc-SC32G_0x490d2b8c-part1
 usb-SanDisk_Cruzer_Glide_3.0_4C530001130415101233-0:0
 usb-SanDisk_Cruzer_Glide_3.0_4C530001130415101233-0:0-part1
 wwn-0x50026b7782008e43
 wwn-0x50026b7782008e43-part1
 wwn-0x50026b7782008e43-part2
 wwn-0x50026b7782008e43-part3
 wwn-0x502b2a201d1c1b1a
 wwn-0x502b2a201d1c1b1a-part1`

2: modify launch_uos.sh add 00:15.0 passthru
 echo "8086 5aa8" > /sys/bus/pci/drivers/pci-stub/new_id
 echo "0000:00:15.0" > /sys/bus/pci/devices/0000:00:15.0/driver/unbind
 echo "0000:00:15.0" > /sys/bus/pci/drivers/pci-stub/bind

-s 16,passthru,0/15/0 \

3. boot uos by
 sh launch_uos.sh -V 1
 root@clr-b175de7560474f7ab7d023f5be776baf ~ # ls /dev/disk/by-id/
ata-@c_A@___E_9__f_____ÃH_c_L___ga
 ata-@c_A@___E_9__f_____ÃH_c_L___ga-part1
 usb-SanDisk_Cruzer_Glide_3.0_4C530001130415101233-0:0
 usb-SanDisk_Cruzer_Glide_3.0_4C530001130415101233-0:0-part1
 virtio-ACRN--67BB-443E-519B
 virtio-ACRN--67BB-443E-519B-part1
 virtio-ACRN--67BB-443E-519B-part2
 virtio-ACRN--67BB-443E-519B-part3
 4. reboot uos
 root@clr-b175de7560474f7ab7d023f5be776baf ~ # ls /dev/disk/by-id/
ata-FORESEE_120GB_SSD_H46092R000318
 ata-FORESEE_120GB_SSD_H46092R000318-part1
 usb-SanDisk_Cruzer_Glide_3.0_4C530001130415101233-0:0
 usb-SanDisk_Cruzer_Glide_3.0_4C530001130415101233-0:0-part1
 virtio-ACRN--67BB-443E-519B
 virtio-ACRN--67BB-443E-519B-part1
 virtio-ACRN--67BB-443E-519B-part2
 virtio-ACRN--67BB-443E-519B-part3
 wwn-0x502b2a201d1c1b1a
 wwn-0x502b2a201d1c1b1a-part1
 sometimes you will get the below info:
 root@clr-b175de7560474f7ab7d023f5be776baf ~ # ls /dev/disk/by-id/
 usb-SanDisk_Cruzer_Glide_3.0_4C530001130415101233-0:0
 usb-SanDisk_Cruzer_Glide_3.0_4C530001130415101233-0:0-part1
*usb-Sony_Storage_459C84140151-0:0
 usb-Sony_Storage_459C84140151-0:0-part1*
 virtio-ACRN--67BB-443E-519B
 virtio-ACRN--67BB-443E-519B-part1
 virtio-ACRN--67BB-443E-519B-part2
 virtio-ACRN--67BB-443E-519B-part3

Expected result:
 The id is same with sos and don't change

Actual result:
 The id is random

mwang106, 2018-12-06:
[External_System_ID] ACRN-1194

mwang106, 2018-12-06:
This bug can not be reproduced in 26550

closed 2018-12-06 by mwang106