projectacrn/acrn-hypervisor #682: Black screens on 4K display when booting with ACRN

kiendinh, 2018-07-12:
I've setup a system with ClearLinux (v23640) and ACRN (master@11e997a7) on the UP2 (APL), and connected a fullHD (1920x1080) monitor and a 4K display. The system has also setup so that Weston launches automatically. When native ClearLinux booted, the two displays showed Weston UI correctly. However, when booting with ACRN to the SOS, both of them only showed black screens. The issue only happened when there was a 4K display. The SOS showed up correctly if both displays were fullHD.

Here is the system setup information:
[UP2 Pentium Quad Core 8GB RAM & 64GB eMMC](https://up-shop.org/up-boards/97-up-squared-pentium-quad-core-8gb-memory64gb-emmc.html)
ClearLinux v23640
acrn-hypervisor (master@ 11e997a7) - built and installed manually - no code modification
ClearLinux kernels:
- kernel-org.clearlinux.native.4.17.5-589
- kernel-org.clearlinux.pk414-sos.4.14.52-63

Please find in the attachment the logs of dmesg and weston service in case of native ClearLinux and in case of SOS.
[weston_4k_sos.log](https://github.com/projectacrn/acrn-hypervisor/files/2187220/weston_4k_sos.log)
[dmesg_4k_sos.log](https://github.com/projectacrn/acrn-hypervisor/files/2187221/dmesg_4k_sos.log)
[weston_4k_native.log](https://github.com/projectacrn/acrn-hypervisor/files/2187222/weston_4k_native.log)
[dmesg_4k_native.log](https://github.com/projectacrn/acrn-hypervisor/files/2187223/dmesg_4k_native.log)

kiendinh, 2018-07-16:
Update: the screens showed Weston UI on a KabyLake NUC. However, the 4K screen only showed 1/4 of the display, which was equivalent to the 1920x1080 resolution of the fullHD screen. The remaining area was black.

kiendinh, 2018-07-18:
Please find in the attachment the weston logs of the KBL-NUC. I found it interesting that 4K display for the UOS, as reported above, still didn't work, but 4K for the SOS did work. Here's how:

The KBL NUC has two display ports: an HDMI and a USB Type-C, which I connected to the display via a USB type-C HDMI adapter. When the USB type-C was connected to 2K monitor, and the HDMI was connected to the 4K monitor, we would have the case reported above: UOS was only showed in 1/4 of the 4K monitor. Here are the weston logs:
[SOS weston log](https://github.com/projectacrn/acrn-hypervisor/files/2205425/SOS_2K_UOS_4K_weston_SOS.log)
[UOS weston log.log](https://github.com/projectacrn/acrn-hypervisor/files/2205426/SOS_2K_UOS_4k_weston_UOS.log)

When the USB type-C was connected to the 4K monitor, and the HDMI was connected to 2K monitor, the SOS was showed correctly @4K resolution, and the UOS was showed correctly @2K resolution. Here are the weston logs
[SOS weston log](https://github.com/projectacrn/acrn-hypervisor/files/2205432/SOS_4K_UOS_2k_weston_SOS.log)
[UOS weston log](https://github.com/projectacrn/acrn-hypervisor/files/2205434/SOS_4K_UOS_2k_weston_UOS.log)

So it seems the order of the monitor connection does affect the results. It's close to display in 4K. In both cases, only part of the 4K monitor information was shared to the UOS - resolutions equal or less than 1600x1200. Is there any restriction in the software which prevents the full info to be passed? Or is it the memory settings like gvt_low_gm_sz... need to be adjusted for higher resolution?

By the way, mouse cursor on UOS in both cases were invisible, though still functioned. It has already reported in [this issue](https://github.com/projectacrn/acrn-hypervisor/issues/706)

Thanks.

Mingyuan18, 2019-03-25:
@kiendinh , May I ask if your question still exists？

kiendinh, 2019-03-25:
@Mingyuan18 I haven't checked it recently. However, since there was no further comments or answers from anyone, I assume that it still exists.

Mingyuan18, 2019-03-25:
In AcrnGT, to support multiple monitors with multiple planes, we have to pre-reserve some resources, which make it impossible to support 4k displays on APL platform.
So it's a known limitation of AcrnGT. Closing it.

gvancuts, 2019-03-25:
There are problems on KBL too, @minhe1 indicated (in https://github.com/projectacrn/acrn-hypervisor/issues/682#issuecomment-405851456) that it may be a software limitation. Let's keep this open to track to a proper resolution (or feature implementation).

Mingyuan18, 2019-03-26:
[External_System_ID] ACRN-3148

Mingyuan18, 2019-06-05:
Due to limited DDB resources in GPU HW, GVT-g cannot support 4k monitor with two monitors and 4 planes on each. So it's a software limitation of GVT-g. 

Suggesting to close it as "Won't Fix”

Mingyuan18, 2019-06-12:
@kiendinh ,Suggesting to close it as "Won't Fix” .

terryzouhao, 2020-04-26:
@fuzhongl  since this is a Won't fix issue/feature, could we close this git issue? 
@kiendinh if you have any further question/issue, feel free submit a new git issue, thanks.

fuzhongl, 2020-05-05:
@kiendinh Please try with gvt-d for 4K display, it is supported since v1.5
Close it as Won't fix because it is the limitation of gvt-g.

closed 2020-05-05 by fuzhongl