projectacrn/acrn-hypervisor #688: Ubuntu build instructions added to device model README?

mattcurfman, 2018-07-12:
Request to include Ubuntu/Debian build instructions added to the device model README

gvancuts, 2018-07-12:
Hi @mattcurfman , we have instructions for Ubuntu/Debian (and Fedora/CentOS too) [here](https://projectacrn.github.io/latest/getting-started/apl-nuc.html#build-acrn-from-source). There are a couple of things @dbkinder and I have in mind regarding this:
* We'll like to move the build instructions into a separate doc (instead of the Getting Started Guide)
* We're trying to centralize as much as possible the documentation in the `doc/` folder (and in that sense, we should really clean up the README.rst in the `devicemodel/` folder to point at the official instructions instead).

Does that make sense to you?

mattcurfman, 2018-07-12:
Yup! 👍

closed 2018-11-15 by mattcurfman