projectacrn/acrn-hypervisor #706: Invisible mouse cursor in UOS

ttzeng, 2018-07-13:
The mouse cursor is invisible in weston and xfce4 running in UOS. I've observed this issue with several UOS versions, my current UOS is pk414-standard.4.14.52-63. I am sure the mouse is detected and functioning, just the cursor is invisible. I also tried to set different `i915.avail_planes_per_pipe` parameter, but nothing changed (the desktop rendered well with invisible mouse cursor). Is `i915.avail_planes_per_pipe` supported? Thanks.

minhe1, 2018-07-18:
@ttzeng  there're several reasons that cursor cannot be displayed:
1. the HW cursor is not yet supported in GVT-g, since it will occupy a HW plane, and GVT-g didn't emulate related registers.
2. the SW cursor require plane restriction feature. You can cat /sys/modules/i915/parameters/avail_planes_per_pipe in UOS to see if the value is correctly set.

Thanks.

ttzeng, 2018-07-18:
Thanks @minhe1, I met the issue on KBL NUC, and it seems the plane restriction feature is only supported on APL? After running the same test on an APL NUC, the mouse cursor is visible now.

minhe1, 2018-07-19:
@ttzeng , plane restriction feature is already supported on KBL NUC. Here's the commited patch:
https://github.com/projectacrn/acrn-kernel/commit/4a791488d3cc9d26e79a01fe8d306d7840469d13#diff-528092536eaeb3c5f7ff6dfeed53786d

closed 2018-08-03 by ttzeng