projectacrn/acrn-hypervisor #707: Issues found with instructions for using Ubuntu as SOS

mattcurfman, 2018-07-13:
- The /etc/grub.d/40_custom file should specify 'insmod ext2', not 'insmod ext4'
- There should be a note that the command line for the kernel in /etc/grub.d/40_custom should be all as a single line, not as multiple lines.  Otherwise the kernel will fail to boot
- There are no instructions for how to enable the requiring tap and networking bridge in the SOS to enable networking in the guest OS
- The link in "Follow the instructions found in the Getting started guide for Intel NUC to install all the build tools and dependencies on your system." points to the wrong URL.  It should point to https://projectacrn.github.io/latest/getting-started/apl-nuc.html#build-acrn-from-source

gvancuts, 2018-07-16:
@ailun258 , can you look at this (I can't assign people to issues directly except for myself).

@dbkinder, regarding the very last point. Is it possible to cross-reference a **section** in an external doc (I have a vague recollection that it's not, although it would be really useful).

ailun258, 2018-07-16:
@mattcurfman  yes, 'insmod ext4' doesn't work here,  should use 'insmod ext2'
i forget if it must be single line, i did it one month ago. 
for the enable the requiring tap and networking under Ubuntu, i created a script, but i don't know where could i push to,

gvancuts, 2018-07-16:
> for the enable the requiring tap and networking under Ubuntu, i created a script, but i don't know where could i push to,

If the script is not too long, I would suggest you add it (as a `code-block`) to the tutorial and instruct people to run this (i.e. copy-paste in a file, save it, `chmod +x` it and then run it).

ailun258, 2018-07-16:
@gvancuts  not too long,

ailun258, 2018-07-16:
@gvancuts        i am generating a new version with latest ACRN and kernel. after complete enabling, i will update this doc to new version later and fix exist problems.

gvancuts, 2018-08-16:
Hi @mattcurfman , #806 has been merged and hopefully addressing all your remarks (latest version is already live here: https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html). Can you review it and close this issue if appropriate?

mattcurfman, 2018-08-16:
Reviewed and looks good.  Thanks @gvancuts, @dbkinder and team.

closed 2018-08-16 by mattcurfman