projectacrn/acrn-hypervisor #910: Graphics Virtualization on RTOS HLD

hwang37, 2018-08-08:
Need Engineering HLD input and architectural feedback. This is the deep dive doc for Graphics Mediator. Would also need to have information on porting the mediator to a different service OS.

hwang37, 2018-08-08:
[External_System_ID]212351

Mingyuan18, 2018-12-19:
AMthis will be delivered as a training/workshop to PAEs in shanghai in April'18, the PPTs would serve as HLDs

closed 2018-12-19 by Mingyuan18

gvancuts, 2018-12-19:
Did you mean April 2018 (in the past) or April 2019 (early next year)?

Can we make those presentations public and available somewhere on the website/wiki?

hwang37, 2018-12-19:
It's about GVT-g for ACRN (a.k.a AcrnGT) design doc and porting guide.  Published in ACRN document website.
https://projectacrn.github.io/latest/developer-guides/hld/hld-APL_GVT-g.html
https://projectacrn.github.io/latest/developer-guides/GVT-g-porting.html

gvancuts, 2018-12-19:
> It's about GVT-g for ACRN (a.k.a AcrnGT) design doc and porting guide. Published in ACRN document website.
> https://projectacrn.github.io/latest/developer-guides/hld/hld-APL_GVT-g.html
> https://projectacrn.github.io/latest/developer-guides/GVT-g-porting.html

Perfect, thanks for the confirmation!