projectacrn/acrn-hypervisor #935: Device: USB support

hwang37, 2018-08-08:
For customer and reference kit configurations which require sharing USB connectivity of different devices to different guest OS, The hypervisor shall enable pass'-through assignment of the physical USB controllers to the Service OS, and the Service OS shall support virtualized device controller support in each guest OS to enable each instance of a physical USB VID:PID device or USB class device to be assigned to a guest OS. (Example only: Any USB mass storage device shall be mapped to virtual controller in IVI, while USB HID device for touch screen assigned to RSE)

hwang37, 2018-08-08:
[External_System_ID]212309

Mingyuan18, 2018-12-20:
AMLatest ACRN github code can support share USB stick, mouse, keyboard.

closed 2018-12-20 by Mingyuan18