projectacrn/acrn-hypervisor #960: CPU: Native HALT support

hwang37, 2018-08-09:
For virtual machines which use a static physical core CPU partitioning, the hypervisor shall pass HALT/MWAIT instructions natively, except for the following exception: in some cases hypervisor may dynamically switch from native execution to trap'-emulation, upon demand of directly enforcing some PM policy, e.g.: 1) turbo mode on one core requires other cores in same package in deep C'-state. It's possible that low'-priority VM doesn't enter deep'-enough C'-state then high'-priority VM cannot acquire desired frequency when requesting turbo. 2) due to some over'-heat condition, hypervisor want to enforce deeper C'-state or lower frequency regardless of guest decision. Above is true even for static partitioning scenario

hwang37, 2018-08-09:
[External_System_ID]211416

Mingyuan18, 2018-12-20:
No regression found in test.

closed 2018-12-20 by Mingyuan18