projectacrn/acrn-hypervisor #1101: missing acrn_mngr.h

shrmrf, 2018-08-24:
I've checked out a fresh copy of this repo and trying to build the device model. My build fails at this include: https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/core/monitor.c#L21

```
core/monitor.c:21:23: fatal error: acrn_mngr.h: No such file or directory
compilation terminated.
Makefile:185: recipe for target '/path/to/acrn-hypervisor/devicemodel/build/core/monitor.o' failed
```

shrmrf, 2018-08-25:
I just compiled the acrn hypervisor and I'm guessing it built the device model, though having the ability to build the DM in isolation would have been better...

closed 2018-08-25 by shrmrf

gvancuts, 2018-08-27:
This is a header file coming from one of the ACRN tool (see tools/acrn-manager). It should be doable to fix the devicemodel Makefile by making acrn_mngr.h a pre-requisite and also include a way to grab it in case it's not available. The limitation (we could live with) would be that we would probably assume that the path to grab the missing header would be `../tools/acrn-manager/`. Again, I don't think this would be a big problem since this is all part of a single git repo that is cloned together.

shrmrf, 2018-08-27:
@gvancuts yeah.. I also figured the same thing out... I closed this issue as it might just become a useless enhancement in the end...