projectacrn/acrn-hypervisor #1115: [APL][IO Mediator][LaaG]Can't receive IOC signal after resume from S3

lishuangx, 2018-08-29:
Environment:
LaaG E2E build:
http://andromeda01.png.intel.com/share/apl/master/20180821-24540/
Image info:
SOS version: 20180821-24540
acrn-hypervisor          2018w34.1.140000p
linux-pk414                 4.14.61     79
SBL IFWI Version： SBL_APL_ACRN_1_0_2_127
IOC Version ：4. 0. 2

Reproduce rate: always

step:
1: make sure IOC&CBC to be enabled, then boot sos
2: in SOS, check IOC environment in sos and cbc_attach normally:
$ ps -aux |grep cbc_attach
   /usr/bin/cbc_attach /dev/ttyS2
3: open signal channel in sos
echo -e "\xFD\x2B\x3C\x4D\c" > /dev/cbc-signals
4: change launch_UOS.sh to append “-i /run/acrn/ioc_$vm_name,0x20 \” and “-l com2,/run/acrn/ioc_$vm_name \”parameters for acrn-dm
5: remove “-s #,virtio-hyper_dmabuf \”  in launch script
6: Run "acrnctl add ./launchUOS.sh –V 1”
7: Run "acrnd &" in sos
9: Check IOC environment in UOS and cbc_attach normally:
   $ ps -aux |grep cbc_attach
      /usr/bin/cbc_attach /dev/ttyS1
10: open signal channel in uos
    echo -e "\xFD\x2B\x3C\x4D\c" > /dev/cbc-signals
11: run test script in uos and can receive signal from sos
  ./ioc_test.sh rx_sig
12: press ignition off and sos&uos suspend
13: press ignition on
14: sos&uos resume, check can continue receive IOC signal in uos

Expected result:
continue receive IOC signal in uos after resume from S3
Actual result:
Can't continue receive signal after resume from S3

lishuangx, 2018-08-29:
[External_System_ID] 252983

mwang106, 2018-09-27:
not reproduce the issue with build below
Image info:
SOS version:           20180918-25060
acrn-hypervisor    2018w38.1.140000p
linux-pk414             4.14.68 99
IFWI Version: ABL_1820HF1
IOC Version ：4. 0. 8
UOS kernel:     c795424a331f [upstream] drm/i915/gvt: Fix stack-out-of-bounds bug in cmd parser

closed 2018-09-27 by mwang106