projectacrn/acrn-hypervisor #1121: SOS hang since UOS does not create ioreq client

mwang106, 2018-08-30:
1. Assign on CPU to  original guest OS
2. Copy ic_vm_start_test to device home directory
3. restart system  and execute ic_vm_start_test
4. Some error logs will be output on SOS terminal(My local error logs is attached, named vhm_ioreq_failed.log, line number may be different based on code base version)

ic_vm_start_test:
ioctl(devfd, IC_CREATE_VM, &create_vm);
ioctl(devfd, IC_SET_IOREQ_BUFFER, (unsigned long)buf);
prepare UOS memory resource...
ioctl(devfd, IC_CREATE_VCPU, &cv);
//Here DO NOT invoke ioctl(devfd, IC_CREATE_IOREQ_CLIENT, 0);
ioctl(devfd, IC_START_VM, &vm_id);

mwang106, 2018-08-30:
[External_System_ID] 237119

mcao6, 2018-08-30:
done

Mingyuan18, 2018-11-19:
below commit fixed this issue, pls close this ticket.
commit: 066adb56fd790e788ff44ac989f9d000bddd196b

closed 2018-11-19 by Mingyuan18