projectacrn/acrn-hypervisor #1131: [graphics] Failed to reach expected results while execute plane restrictions tests.

xinyunliu, 2018-08-31:
plane restrictions test cases cannot reach expected results.   UOS has  no show as expected with this setting:
 * SOS: i915.domain_plane_owner=0x11111110000
 * UOS: i915.domain_plane_owner=0x000F00

Steps
--------

1.  Build sos_boot.img with `i915.domain_plane_owner=0x11111110000` , `i915.avail_planes_per_pipe=0x00000F` (pipe-A for Dom0,pipe-B for LaaG)
    a. change parameter in acrn-ebtool/acrn-hypervisor/devicemodel/samples/apl-mrb/sos_bootargs_release.txt or sos_bootargs_debug.txt(depends on debug or release build version)
    b. make sos_boot  (sos_boot.image will generate in pub, BTW, make stitch then make sos_boot maybe the easy way to make this)
2. Flash SOS to MRB only 
3. Connect display then restart
4. Boot system, enter Dom0
5. In SOS,edit launch_uos.sh, modify below row
```
    echo "Launch clearlinux UOS"
    launch_clearlinux 1 3 "64 448 8" 0x000F00 clearlinux 0
```
6. Make sure the parameter you modified in sos_bootargs_release.txt and launch_UOS.sh are work
   `cat /proc/cmdline` for checking
7. Observe the display appearance

Current result:
--------------
2 monitors with UOS:  HDMI-1 display SOS, HDMI-2 no display;

Expected result:
--------------
HDMI-1 display SOS, HDMI-2 UOS display;

xinyunliu, 2018-08-31:
251520

dandanyx, 2018-09-25:
Cannot reproduce this issue on WW38 LaaG E2E build.
Please help to close it,thanks!

xinyunliu, 2018-09-25:
closed as integrated

closed 2018-09-25 by xinyunliu

mwang106, 2018-09-26:
SOS version:    20180918-25060
acrn-hypervisor:  2018w38.1.140000p
linux-pk414: 4.14.68 99
IFWI Version: ABL_1820HF1
IOC Version: 4.0.8
UOS kernel:  c795424a331f [upstream] drm/i915/gvt: Fix stack-out-of-bounds bug in cmd parser