projectacrn/acrn-hypervisor #1133: /usr/bin/ld: /home/charleye/acrn-hypervisor/build/hypervisor/boot/reloc.o: relocation R_X86_64_32S against `.text.get_hv_image_delta' can not be used when making a shared object; recompile with -fPIC /home/charleye/acrn-hypervisor/build/hypervisor/boot/reloc.o: error adding symbols: Bad value

chasinglulu, 2018-08-31:
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.5-4ubuntu2' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.5 (Ubuntu 4.8.5-4ubuntu2)

gvancuts, 2018-09-25:
@chasinglulu @kiendinh can you confirm what exactly fixed the issue for you?

I would like to know if we need to amend our "Getting Started Guide" with additional instructions (or a note): https://projectacrn.github.io/latest/getting-started/apl-nuc.html#build-acrn-from-source

kiendinh, 2018-09-25:
@gvancuts, I used to checkout the tag which was known to compile (ie, acrn-2018w35.4-140000p). However today the latest master branch was built successfully on Ubuntu 16.04 with binutils 2.26. Some recent changes may have fixed the issue.

gvancuts, 2018-09-26:
> @gvancuts, I used to checkout the tag which was known to compile (ie, acrn-2018w35.4-140000p). However today the latest master branch was built successfully on Ubuntu 16.04 with binutils 2.26. Some recent changes may have fixed the issue.

Thanks Kien. No need to modify the existing doc then!

gvancuts, 2019-04-24:
I'm re-opening this issue as using the latest `binutils` (2.26) version from the Ubuntu 16.04 repo does not work for me. Manually upgrading to 2.27 (or possibly higher) is required as far as I can tell.

gvancuts, 2019-06-11:
Closing this issue as upgrading `binutils` to 2.27 solves the build issue on Ubuntu 16.04

closed 2019-06-11 by gvancuts