projectacrn/acrn-hypervisor #1135: Clear Linux image not longer available on the download website

gvancuts, 2018-08-31:
The Clear Linux image that we use as a [reference](https://download.clearlinux.org/releases/23690/clear/clear-23690-installer.img.xz) in our [Getting Started Guide](https://projectacrn.github.io/latest/getting-started/apl-nuc.html#set-up-a-clear-linux-operating-system) is no longer available from the download website. This is a new policy from the Clear Linux team. We need to update the instructions to use a newer (ideally the latest as of today) image as a reference.

gvancuts, 2018-08-31:
@ttzeng @kiendinh @ailun258 is this something that one of you could do?

@dbkinder , FYI

dbkinder, 2018-08-31:
We could use instructions that don't go obsolete within a day as new releases of CL are made :(

gvancuts, 2018-08-31:
> We could use instructions that don't go obsolete within a day as new releases of CL are made :(

Couldn't agree more.. the path to that in my opinion is a bundle that *really* installs all components where they ought to be (and configure the system accordingly). That would also solve the "upgrade" issue where one cannot upgrade the Service OS at all without breaking the installation.

dbkinder, 2018-08-31:
So what's stopping such a bundle from being implemented?  Is this a CL or ACRN issue (or both)?

gvancuts, 2018-08-31:
So what's stopping such a bundle from being implemented? Is this a CL or ACRN issue (or both)?

Proper packaging for ACRN isn’t straightforward, nothing that cannot be solved I’m sure but it would take a bit of time that nobody has really invested in yet. I’m not sure whose job it is to solve this.

One aspect is the ESP manipulation (and its equivalent when using SBL) and the other one is the platform detection (apl-mrb vs. NUC, EFI vs. SBL, etc). That’s for the SOS part of it. Then there is a similar challenge when dealing with Clear as a UOS.

Mingyuan18, 2019-04-08:
@gvancuts , What is the current situation of this bug, do we still need to track?

gvancuts, 2019-04-08:
This one will come and go, depending on how long images will be posted on Clear Linux download website and how often we release new versions. The instructions are a lot more generic than they used to be, I think we can close this ticket.

closed 2019-04-08 by gvancuts