projectacrn/acrn-hypervisor #1159: [APL][IO Mediator][AaaG]OTG VBUS signal does not support in IOC mediator whitelist

liuyuan-GitHub, 2018-09-05:
Environment:
LaaG E2E build:
http://andromeda01.png.intel.com/share/apl/master/20180821-24540/
Image info:
SOS version: 20180821-24540
acrn-hypervisor 2018w34.1.140000p
linux-pk414 4.14.61 79
SBL IFWI Version： SBL_APL_ACRN_1_0_2_127
IOC Version ：4. 0. 8

Signal test
Signal ID: ias_cbc_tx_signal_id_IasUsbOtgVbusCtrl = 20028U(0x4E3C)
Signal Length: 1 bit.
Signal value: 0 – reset USB OTG VBUS power, 1 – set USB OTG VBUS power

liuyuan-GitHub, 2018-09-05:
[External_System_ID] 253436

mwang106, 2018-09-18:
Verified On

AaaG build and AaaG weekly build:
Image info:
SOS version:             20180914-25010
acrn-hypervisor:    2018w37.4.140000p
linux-pk414:             4.14.68          98
IFWI Version: ABL_1820HF1
IOC Version ：4. 0. 8

Not reproduce the issue

closed 2018-09-18 by mwang106