projectacrn/acrn-hypervisor #1193: HDMI and VGA monitors on NUC  have no display after boot up SOS and start weston in SOS.

dandanyx, 2018-09-07:
Environment:
------------
SOS kernel: 4.14.67-92.pk414-sos 
UOS kernel: 4.14.67-92.pk414-standard 
Clearlinux Version: 24870
Hardware: NUC

Bug detailed description:
-------------
HDMI and VGA monitors have no display after boot up SOS and start weston in SOS.

Reproduce steps:
-------------
1. Boot up SOS
2. Execute the command "weston --tty=1 -i 0 &" to start weston in SOS.

Current result:
--------------
HDMI and VGA monitors have no display after boot up SOS and start weston in SOS.

Expected result:
--------------
HDMI and VGA monitors can display SOS mosaic screen after boot up SOS and start weston in SOS.

Basic root-causing log:
--------------
no log

mwang106, 2018-09-07:
[External_System_ID] 253878

mwang106, 2018-09-18:
Verified on
NUC WW38.1 daily build: https://download.clearlinux.org/releases/25040/
Image info:
acrn-hypervisor: 2018w37.5.140000p
linux-pk414: 4.14.68 98
Cannot reproduce this issue.

closed 2018-09-18 by mwang106