projectacrn/acrn-hypervisor #1208: Failed to create new gvt instance when system can not alloc 2M contiguous pages

zhipenggong, 2018-09-11:
In some scenarios kernel can not alloc 2M contiguous pages for ggtt pv and GVT-g fails to create new intance.

[22968.459987] mevent: page allocation failure: order:9, mode:0x14000c0(GFP_KERNEL), 
nodemask=(null)                                                                                              [1252/437803]
[22968.459999] mevent cpuset=/ mems_allowed=0
[22968.460009] CPU: 0 PID: 392 Comm: mevent Tainted: G     U          4.14.67-sos+ #2
[22968.460011] Call Trace:
[22968.460025]  ? dump_stack+0x85/0xc5
[22968.460030]  ? warn_alloc+0xe0/0x170
[22968.460034]  ? __alloc_pages_slowpath+0xe21/0xe60
[22968.460036]  ? get_page_from_freelist+0x28d/0xae0
[22968.460041]  ? sg_init_table+0x15/0x70
[22968.460044]  ? __alloc_pages_nodemask+0x255/0x270
[22968.460049]  ? gen8_mm_alloc_page_table+0xd0/0x490
[22968.460051]  ? intel_vgpu_create_mm+0x101/0x280
[22968.460054]  ? intel_vgpu_init_gtt+0xa9/0x530
[22968.460057]  ? intel_gvt_create_vgpu+0x160/0x250
[22968.460062]  ? acrngt_instance_create+0x5c/0x290
[22968.460064]  ? acrngt_sysfs_add_instance+0x8f/0x180
[22968.460067]  ? acrngt_sysfs_instance_manage+0xcf/0x1c0
[22968.460069]  ? acrngt_sysfs_instance_manage+0xcf/0x1c0
[22968.460073]  ? x86_pmu_cancel_txn+0x3c/0xa0
[22968.460077]  ? kernfs_fop_write+0x10b/0x190
[22968.460081]  ? __vfs_write+0x26/0x140
[22968.460084]  ? selinux_file_permission+0x116/0x130
[22968.460087]  ? __sb_start_write+0xa5/0xe0
[22968.460089]  ? vfs_write+0xbe/0x1b0
[22968.460091]  ? SyS_write+0x45/0xa0
[22968.460093]  ? do_syscall_64+0x6f/0x130
[22968.460097]  ? entry_SYSCALL_64_after_hwframe+0x42/0xb7

mwang106, 2018-09-17:
[External_System_ID] 255249

dandanyx, 2018-10-16:
On WW42 AaaG build, cannot reproduce this issue after reboot AaaG for 150 times continually.
Suggest to close this issue, thanks!

closed 2018-10-16 by mwang106