projectacrn/acrn-hypervisor #1234: Build failed on Ubuntu 16.04

kiendinh, 2018-09-13:
acrn-hypervisor (master branch @ c6b7940b) is failed to build on Ubuntu 16.04.
See [build.log](https://github.com/projectacrn/acrn-hypervisor/files/2378047/build.log)

It was built successfully @ acrn-2018w35.4-140000p but failed from acrn-2018w35.5-140000p. Same error. So some commits between these two tags have caused the issue. Please check.

gvancuts, 2018-09-13:
It looks the same as what was reported in #1133 , can you check it out?

kiendinh, 2018-09-14:
Yes, I've confirmed they are the same. Closing.

closed 2018-09-14 by kiendinh

Vshmuk, 2019-01-31:
Now it says:

/usr/bin/ld: warning: -z noreloc-overflow ignored.
/usr/bin/ld: /home/xxx/projects/acrn-hypervisor/build/hypervisor/boot/reloc.o: relocation R_X86_64_32S against `.text.get_hv_image_delta' can not be used when making a shared object; recompile with -fPIC
/home/ebeysembaev/projects/acrn-hypervisor/build/hypervisor/boot/reloc.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status

Something is broken in my setup?