projectacrn/acrn-hypervisor #1242: APL][IO Mediator][LaaG&AaaG]USB mediator Mode test equipment needs to be plugged in for identification

mwang106, 2018-09-14:
Environment:
LaaG E2E build:
Image info:
SOS version: 20180813-24360
acrn-hypervisor          2018w32.5.140000p
linux-pk414                 4.14.61     76
SBL IFWI Version： SBL_APL_ACRN_1_0_2_114
IOC Version ：4. 0. 2

step:
1:insert USB stick in standard USB port
2:boot uos by sharing the  USB stick in cmd line
-s n,xhci,1-1:1-2:1-3:2-1:2-2:2-3 \
3:uos can access USB mass storage.

Expected result:
uos can access USB mass storage.

Actual result:
uos can't access USB storage,need to be plugged in

mwang106, 2018-09-14:
[External_System_ID] 252187

Mingyuan18, 2018-11-19:
insert multiple devices in standard port shared by xhci, boot up uos, do not plugged in can find the devices directly.

closed 2018-11-19 by Mingyuan18