projectacrn/acrn-hypervisor #1243: [APL][IO Mediator][Laag]Touch with build in hub can not work in UOS during USB mediator test

mwang106, 2018-09-14:
•Environment:
•HW: MRB D0 Premium
•Build:
LaaG E2E build：
Image info:
SOS version:           20180908-24920
acrn-hypervisor     2018w36.5.140000p
linux-pk414             4.14.67 94
SBL IFWI Version： SBL_APL_ACRN_1_0_2_127
IOC Version ：4. 0. 7

•Reproduce rate:  100% 
•Reproduce steps:

•1.connect touch screen with build in hub and boot to sos
•2. remove USB pssthru related parameters in launch script
3. boot UOS with USB mediator in launch script:
    -s n,xhci,1-1:1-2:1-3:2-1:2-2:2-3
4. in laag and aaag to check the touch with build in hub work normally.
•Expected result:  the touch with build in hub work normally in aaag and laag with usb mediator
•Actual result: USB mediator test on touch with build in hub not work in laag&aaag
•Debugging info:

mwang106, 2018-09-14:
[External_System_ID] 255010

Mingyuan18, 2018-11-19:
Not reproduce the issue with this weekly build

closed 2018-11-19 by Mingyuan18