projectacrn/acrn-hypervisor #1254: Replace unbounded functions with safe string and buffer handling  functions in ACRN DM & Tools

yonghuah, 2018-09-17:
During static code analysis of the ACRN hypervisor & kernel source code, excessive counts of unsafe functions were found in use. Use of deprecated and dangerous functions is not recommended as they often have known security weaknesses, leading to
resultant weaknesses such as buffer overflows and other unpredictable behavior. Unsafe functions also allows the attackers to exploit software through various binary-level attacks

mwang106, 2018-09-17:
[External_System_ID] 254443

mwang106, 2019-01-11:
all banned functions  have been removed for device model and tools modules. and also we refined the usage for functions that can be used conditionally.

closed 2019-01-11 by mwang106