projectacrn/acrn-hypervisor #1262: acrnctl tool should not only gives RC 0

LibertadCruz, 2018-09-17:
I have notice that the acrnctl tool even if it fails to execute the command it returns 0. This is missleading if you are automating and you want to relly on a RC 1 for error.

It would be nice if the acrnctl tool gave RC 1 when it failed.

lyan3, 2018-12-28:
Thanks for reporting the issue, we kept the issue pending because there is a discussion to remove usage of all system() calls. Once there is a decision, we will move forward.

Mingyuan18, 2018-12-28:
[External_System_ID] ACRN-2358

Mingyuan18, 2019-04-11:
@LibertadCruz ,The patch for this problem has been merged. Can you reproduce this problem now, if not, please close this issue. Thanks!

Mingyuan18, 2019-04-16:
This pr's acrn tag is : acrn-2019w14.1-140000p

Mingyuan18, 2019-05-08:
@LibertadCruz  Please verify this issue , if pass , please close it.

fuzhongl, 2019-05-24:
@LibertadCruz Patch was merged. Could you please close this ticket?
Thanks!

fuzhongl, 2019-05-30:
@LibertadCruz 
Patch was merged in CL 28600, please help to verify this issue; if fixed, please close this issue.
Any more concern, please update the comments. If no more feedback within two weeks, this ticket will be closed by default. 
You can reopen it if this issue can be reproduced for new release.
Appreciated for your understanding.

Mingyuan18, 2019-06-11:
Close this issue due to no any feedback.

closed 2019-06-11 by Mingyuan18