projectacrn/acrn-hypervisor #1269: vioapic: change the method for assert/deassert

JasonChenCJ, 2018-09-18:
a guest may use per-cpu vector allocation mechanism which may lead to
    same vector on different vcpu for different vioapic pins.
    when we broadcast EOI from vlapic to vioapic, it could trigger
    ptdev_intx_ack for incorrect vioapic pin which will lead to incorrect
    vioapic_irq_deassert.

    old implementation is recording assert & deassert by acnt, like:
    - assert -> acnt++
    - deassert -> acnt--
    which means an incorrect deassert may descrease acnt from 0 to -1, it leads
    to error.

mwang106, 2018-09-25:
[External_System_ID] 255851

ftangx, 2018-11-26:
Defect 255851 has verified.

Mingyuan18, 2018-11-29:
Can't reproduce this issue in SOS:260120

closed 2018-11-29 by Mingyuan18