projectacrn/acrn-hypervisor #1277: fastboot reboot cause acrn-dm crash

mwang106, 2018-09-18:
Steps to reproduce:
1.Boot to UOS
2. Input command:
   # adb reboot bootloader
   # fastboot reboot
  or
   # fastboot reboot-bootloader

Logs:
USB for fastboot transport layer selected
[2018-09-17_11h48.51.230802]

GOT getvar:slot-count
[2018-09-17_11h48.51.231458]

SENT OKAY2
[2018-09-17_11h48.51.232702]

GOT reboot-bootloader
[2018-09-17_11h48.51.236443]

Rebooting to bootloader ...
[2018-09-17_11h48.51.237481]

SENT OKAY
[2018-09-17_11h48.51.238521]

full reset
[2018-09-17_11h48.51.248717]
attach ioreq client return 1 (1 = destroying, could be triggered by Power State change, others = error)
[2018-09-17_11h48.51.250500]
VM loop exit
[2018-09-17_11h48.51.355618]
acrn-dm: hw/pci/core.c:496: modify_bar_registration: Assertion `error == 0' failed.
[2018-09-17_11h48.51.975467]
/usr/share/acrn/samples/apl-mrb/launch_uos.sh: line 133:   431 Aborted                 (core dumped) acrn-dm -A -m $mem_size -c $2$boot_GVT_option"$GVT_args" -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio $npk_virt -s 9,virtio-net,$tap -s 3,virtio-blk$boot_dev_flag,/data/$5/$5.img -s 7,xhci,1-1:1-2:1-3:2-1:2-2:2-3:cap=apl -s 8,passthru,0/15/1 -s 13,virtio-rpmb -s 10,virtio-hyper_dmabuf -s 11,wdt-i6300esb -s 14,passthru,0/e/0,keep_gsi -s 23,passthru,0/17/0 -s 15,passthru,0/f/0 -s 27,passthru,0/1b/0 -s 24,passthru,0/18/0 -s 18,passthru,3/0/0,keep_gsi $boot_ipu_option -i /run/acrn/ioc_$vm_name,0x20 -l com2,/run/acrn/ioc_$vm_name $boot_image_option --enable_trusty -B "$kernel_cmdline" $vm_name

mwang106, 2018-09-18:
[External_System_ID] 255427

Mingyuan18, 2018-11-19:
Can not reproduce this by bug by using latest build-ww43 weekly build

closed 2018-11-19 by Mingyuan18