projectacrn/acrn-hypervisor #1286: There are warnings when use allyes config to compile

liuwei987, 2018-09-19:
There are warnings when use allyes config to compile:
drivers/char/vhm/vhm_dev.c: In function ‘vhm_dev_ioctl’:
./include/linux/kern_levels.h:5:18: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘long int’ [-Wformat=]
#define KERN_SOH "\001" /* ASCII Start Of Header /
^
./include/linux/kern_levels.h:11:18: note: in expansion of macro ‘KERN_SOH’
#define KERN_ERR KERN_SOH "3" / error conditions */
^~~~~~~~
./include/linux/printk.h:306:9: note: in expansion of macro ‘KERN_ERR’
printk(KERN_ERR pr_fmt(fmt), ##VA_ARGS)
^~~~~~~~
drivers/char/vhm/vhm_dev.c:645:4: note: in expansion of macro ‘pr_err’
pr_err("vhm-dev: monitor intr data err=%d\n", ret);
^~~~~~
drivers/char/vhm/vhm_dev.c:645:44: note: format string is defined here
pr_err("vhm-dev: monitor intr data err=%d\n", ret);
^
%ld
In file included from ./include/linux/printk.h:334:0,
from ./include/linux/kernel.h:14,
from drivers/dma-buf/hyper_dmabuf/hyper_dmabuf_msg.c:29:
drivers/dma-buf/hyper_dmabuf/hyper_dmabuf_msg.c: In function ‘cmd_process_work’:
drivers/dma-buf/hyper_dmabuf/hyper_dmabuf_msg.c:220:29: warning: format ‘%lx’ expects argument of type ‘long unsigned int’, but argument 4 has type ‘long long unsigned int’ [-Wformat=]
dev_dbg(hy_drv_priv->dev, "\tgrefid 0x%lx\n",
^
(u64)req->op[8] << 32 | req->op[7]);
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/dynamic_debug.h:135:39: note: in definition of macro ‘dynamic_dev_dbg’
__dynamic_dev_dbg(&descriptor, dev, fmt, 
^
drivers/dma-buf/hyper_dmabuf/hyper_dmabuf_msg.c:220:3: note: in expansion of macro ‘dev_dbg’
dev_dbg(hy_drv_priv->dev, "\tgrefid 0x%lx\n",

mwang106, 2018-09-19:
[External_System_ID] 255496

liuwei987, 2018-09-20:
verified. closed it.

mwang106, 2018-09-27:
reopen due to still exist warning in gvt related this issue

closed 2018-09-29 by mwang106